veista/nilan #1: Unable to add integration

abstacek, 2022-11-22:
[image]
Unable to add integration. Is it necessary to edit configuration.yaml manually, I just don't know how?

veista, 2022-11-22:
Hi, the issue is now resolved with the latest release.

abstacek, 2022-11-23:
Yes already ok, unsupported device for me unfortunately :(

veista, 2022-11-23:
Which device do you have? If it is cts602 based adding support is rather easy.

abstacek, 2022-11-23:
I have a Combi 302 polar top. Can you please advise how to edit/proceed?

abstacek, 2022-11-23:
Hello, could you please advise me how to add support for the Nilan Combi
302 polar? I'll be happy to pay you for your work, I don't want anything
for free. Thank you very much for your help. Adam

veista, 2022-11-23:
Ok so first you need to install the newest version 1.0.3. Then take a picture of the **type plate** of your device and add it here.
Then turn on debug logging of the integration in your configuration.yaml file like:
`logger:
  default: info
  logs:
    custom_components.nilan: debug`

Report the debug string here

If you want to support me, you can donate here https://www.buymeacoffee.com/veista

veista, 2022-11-24:
Thanks, do you have the HMI?

abstacek, 2022-11-24:
Yes I do, do you mind if it's not in english?
[image]

veista, 2022-11-24:
Thanks, no this is great. I will add this ASAP.

veista, 2022-11-24:
Hi,

The release is now available for you to test. It is beta releases, so make sure you enable beta releases if you are using HACS.

Please let me know if something is missing or something doesn't work.

abstacek, 2022-11-24:
Hi, it worked :-) Great job. Showing 55 entities.
[image]

veista, 2022-11-24:
That's great! I added a bit better support for your device on version 1.0.5B. Please let me know how it works!

veista, 2022-11-24:
Also do you have an after-heating element installed on your device?

abstacek, 2022-11-24:
The new version 1.0.5b works. I have the polar version with preheating of the supply air so that the exchanger in the unit does not freeze.

veista, 2022-11-24:
[image]

I mean an accessory, you can see it in the HMI menu.

abstacek, 2022-11-24:
This item is in the menu. I don't know if it is actually installed.
[image]

abstacek, 2022-11-24:
[image]

veista, 2022-11-24:
Great! Then I will code the support for after heating and you can test it.

abstacek, 2022-11-24:
OK. I will try.

veista, 2022-11-25:
So i released the 1.0.6B last night. If you could go through the hmi settings and cross-check that every value changes where it is supposed to and also that there is no unnecessary entities.

abstacek, 2022-11-25:
Hi, I ran into a problem. HA and nilan integration stops working after about a minute. It's like it stopped communicating via modbus. I tried communicating via node-red and it works. I use USR-TCP232-410S for communication, I don't know where the error could be?

veista, 2022-11-25:
Is there anything in the logs?

abstacek, 2022-11-25:
Yes, for example,:
Updating nilan binary_sensor took longer than the scheduled update interval 0:00:10

abstacek, 2022-11-25:
2022-11-25 19:58:52.381 WARNING (MainThread) [homeassistant.components.number] Updating nilan number took longer than the scheduled update interval 0:00:10
2022-11-25 19:58:52.467 WARNING (MainThread) [homeassistant.components.select] Updating nilan select took longer than the scheduled update interval 0:00:10
2022-11-25 19:58:52.647 WARNING (MainThread) [homeassistant.components.climate] Updating nilan climate took longer than the scheduled update interval 0:00:10

abstacek, 2022-11-25:
Only restarting the home assistant will help.

veista, 2022-11-25:
Did you try if reloading the integration helps? Did this problem arise with the latest release?

abstacek, 2022-11-25:
Yes, reload integration will help. But in few minutes it is unavailable again. In previous versions it did too, but I was constantly doing something in HA and restarting HA so I didn't pay much attention to it.

abstacek, 2022-11-25:
ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.nilan_time fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 527, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 722, in async_device_update
    raise exc
  File "/config/custom_components/nilan/sensor.py", line 397, in async_update
    self._attr_native_value = await getattr(self._device, self._attribute)()
  File "/config/custom_components/nilan/device.py", line 1807, in get_time
    result.registers[i].to_bytes(2, "little", signed=False),
IndexError: list index out of range

veista, 2022-11-25:
It might be an issue with the bridge then. I'm afraid I can only guess from here, but the error comes from that it does not get a response from the device.

abstacek, 2022-11-25:
Yes, there could be a fault with the bridge. I don't really understand that. I just find it strange that even if HA stops communicating, node-red and modbus communicate. So the bridge is communicating, it's just not communicating with HA.

veista, 2022-11-25:
Please note that with these bridges usually, you can only communicate with one TCP client at a time, especially if you run two programs on the same computer, you will run into issues since the server(bridge) does not know which client to respond to and then they get confused.

abstacek, 2022-11-25:
I'm sorry, I didn't know that.

veista, 2022-11-25:
Yes, try to disconnect all other clients from the bridge so that you only have the HA on that bridge. You need another bridge if you want more devices on the RTU network.

abstacek, 2022-11-25:
I have only the nilan combi 302 connected to the modbus bridge and only HA turned on. But still the same problem.

veista, 2022-11-25:
Ok. Then I guess you can try modifying the scan interval or try disabling entities. If you have vs code addon you can modify them there or via text editor if you have samba share.

veista, 2022-11-25:
You can also turn the debug log on for homeassistant.components.modbus if this brings any new info.

abstacek, 2022-11-25:
I will try to adjust the polling interval. This is my bridge setup. Nilan still mentions the use of a 120ohm resistor in the manual, so I'll try to find that too. It is possible that my bridge is not compatible with nilan.
[image]

veista, 2022-11-25:
I'm not sure what the issue is, but i don't think it's the resistor. Did you turn the modbus debugging on in HA?

veista, 2022-11-25:
Can you disable that sync baudrate option and try that?

abstacek, 2022-11-25:
Yes.

2022-11-25 23:17:31.075 INFO (MainThread) [homeassistant.setup] Setup of domain input_text took 0.5 seconds
2022-11-25 23:17:31.076 INFO (MainThread) [homeassistant.setup] Setup of domain input_datetime took 0.5 seconds
2022-11-25 23:17:31.076 INFO (MainThread) [homeassistant.setup] Setup of domain zone took 0.5 seconds
2022-11-25 23:17:31.091 INFO (SyncWorker_4) [homeassistant.components.modbus.modbus] modbus Nilan communication open
2022-11-25 23:17:31.097 INFO (MainThread) [homeassistant.setup] Setting up hardware
2022-11-25 23:17:31.126 INFO (MainThread) [homeassistant.setup] Setup of domain input_select took 0.6 seconds
2022-11-25 23:17:31.133 INFO (MainThread) [homeassistant.setup] Setup of domain input_number took 0.6 seconds
2022-11-25 23:17:31.138 INFO (MainThread) [homeassistant.setup] Setup of domain schedule took 0.6 seconds
2022-11-25 23:17:31.148 INFO (MainThread) [homeassistant.setup] Setup of domain hardware took 0.1 seconds
2022-11-25 23:17:31.153 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.apex
2022-11-25 23:17:31.225 INFO (MainThread) [homeassistant.setup] Setting up hacs
2022-11-25 23:17:31.225 INFO (MainThread) [homeassistant.setup] Setup of domain hacs took 0.0 seconds
2022-11-25 23:17:31.235 INFO (MainThread) [homeassistant.setup] Setting up met
2022-11-25 23:17:31.235 INFO (MainThread) [homeassistant.setup] Setup of domain met took 0.0 seconds
2022-11-25 23:17:31.249 INFO (SyncWorker_2) [homeassistant.loader] Loaded switch from homeassistant.components.switch
2022-11-25 23:17:31.255 INFO (MainThread) [homeassistant.setup] Setup of domain input_button took 0.7 seconds
2022-11-25 23:17:31.256 INFO (MainThread) [custom_components.hacs]

abstacek, 2022-11-25:
Is your modbus integration different from other modbus integrations? Sorry for the questions, but I'm not an expert on this.

veista, 2022-11-25:
No. I use the HA built-in modbus library with pretty much default modbus TCP values. It might be that the bridge you are using is not using modbus TCP but only forwards modbus RTU messages. This would mean that it would need a framer, which the HA modbus library has  but it would mean some more coding. I wish I could test that is this the case.

abstacek, 2022-11-25:
Ah, I confess that I do not understand it at all. I would be happy if it worked, I was glad that you modified your program for the Combi 302 and I did not expect that there would be a problem with the bridge. Will I have to buy a new bridge?

abstacek, 2022-11-25:
I turned off sync baudrate but same problem :(

veista, 2022-11-25:
So yes a quick test. In device.py file whre it says "tcp" replace with "rtuovertcp"

abstacek, 2022-11-25:
2022-11-26 00:19:10.911 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: nilan
[image]
after reboot, initialization is in progress

abstacek, 2022-11-25:
Do you want access to HA and bridge?

veista, 2022-11-25:
No, I don't think that would be of any value. It's what I thought, but had to try. I guess you can try increasing the timeout and other settings at the same section, where the modbus config is and see if something works.

abstacek, 2022-11-25:
Ok, I'll try. Thank you very much for your time. I really appreciate it.

veista, 2022-11-25:
Add "message_wait_milliseconds": 30, line after "delay": "",

abstacek, 2022-11-25:
"delay": "", "message_wait_milliseconds": 30,

correctly?

veista, 2022-11-25:
Newline after delay

abstacek, 2022-11-26:
It communicates for 3 minutes and then nothing.

2022-11-26 01:03:07.714 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_return_fan_power_at_level_2 is taking over 10 seconds
2022-11-26 01:03:07.715 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_return_fan_power_at_level_3 is taking over 10 seconds
2022-11-26 01:03:07.716 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_return_fan_power_at_level_4 is taking over 10 seconds
2022-11-26 01:03:07.716 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_fan_start_up_delay is taking over 10 seconds
2022-11-26 01:03:07.716 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_defrost_start_setpoint is taking over 10 seconds
2022-11-26 01:03:07.717 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_defrost_stop_setpoint is taking over 10 seconds
2022-11-26 01:03:07.718 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_compressor_stop_time is taking over 10 seconds
2022-11-26 01:03:07.719 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_low_room_temperature_setpoint is taking over 10 seconds
2022-11-26 01:03:07.719 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_minimum_supply_air_temperature_in_summer is taking over 10 seconds
2022-11-26 01:03:07.720 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_maximum_supply_air_temperature_in_summer is taking over 10 seconds
2022-11-26 01:03:07.720 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.nilan_room_temperature_neutral_zone is taking over 10 seconds
2022-11-26 01:03:07.747 WARNING (MainThread) [homeassistant.helpers.service] Unable to find referenced entities input_boolean.lenka_doma or it is/they are currently not available
2022-11-26 01:03:07.924 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_air_filter_alarm_interval is taking over 10 seconds
2022-11-26 01:03:07.925 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_indoor_air_quality_control is taking over 10 seconds
2022-11-26 01:03:07.926 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_cooling_mode_ventilation_level is taking over 10 seconds
2022-11-26 01:03:07.926 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_cooling_setpoint_offset is taking over 10 seconds
2022-11-26 01:03:07.928 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_low_humidity_ventilation_level is taking over 10 seconds
2022-11-26 01:03:07.928 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_high_humidity_ventilation_level is taking over 10 seconds
2022-11-26 01:03:07.929 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_hmi_language is taking over 10 seconds
2022-11-26 01:03:07.929 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_low_outdoor_temp_ventilation_level is taking over 10 seconds
2022-11-26 01:03:07.929 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_minimum_ventilation_supply_level is taking over 10 seconds
2022-11-26 01:03:07.930 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_minimum_ventilation_return_level is taking over 10 seconds
2022-11-26 01:03:07.930 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_maximum_ventilation_return_level is taking over 10 seconds
2022-11-26 01:03:07.931 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_defrost_ventilation_level is taking over 10 seconds
2022-11-26 01:03:07.931 WARNING (MainThread) [homeassistant.helpers.entity] Update of select.nilan_reset_alarm is taking over 10 seconds
2022-11-26 01:03:08.334 WARNING (MainThread) [homeassistant.helpers.entity] Update of climate.nilan_hvac is taking over 10 seconds
2022-11-26 01:03:12.714 WARNING (MainThread) [homeassistant.components.number] Updating nilan number took longer than the scheduled update interval 0:00:15
2022-11-26 01:03:12.924 WARNING (MainThread) [homeassistant.components.select] Updating nilan select took longer than the scheduled update interval 0:00:15
2022-11-26 01:03:13.334 WARNING (MainThread) [homeassistant.components.climate] Updating nilan climate took longer than the scheduled update interval 0:00:15

veista, 2022-11-26:
Set the timeout delay to 1 there is a typo it is 1000. Thats at least one issue I found in the settings.

abstacek, 2022-11-26:
That's it! :-) It's been working for over 10 minutes without a problem. I believe the timeout was causing the whole problem.

veista, 2022-11-26:
Yes the issue was that your bridge doesnt send a response to some message and it was waiting for 1000seconds for that message instead of rejecting the request after 1s and retrying. It was  1000s is longer than the 10s interval so thats why HA couldn't recover.

abstacek, 2022-11-26:
But I'm glad you figured out the problem and were able to fix it. You implemented the supply heating function as you asked about it?

veista, 2022-11-26:
I added the config settings, yes. So if you can cross check the against the HMI it would be great. So change the value in HMI and look in HA if it changes and reads the same value and vice versa.

abstacek, 2022-11-26:
Hello, would it be possible to add these entities (sensors)?
Set temperature
suction fan speed
exhaust fan speed

veista, 2022-11-26:
They should be there. Some options are nested inside the climate entity.

[30 more replies]

closed 2022-11-26 by veista