veista/nilan #3: Time betw.de-icing

abstacek, 2022-11-28:
Please add Time between defrosts. This option is available in the HMI. Thank you.

abstacek, 2022-11-30:
Hi, has the time between defrosts been added yet? Thank you.

veista, 2022-11-30:
No. I will add it when I have time, there are some issues in my code since I havn't prepared for it. My device doesn't have those and there is a naming conflict.

abstacek, 2022-11-30:
OK

veista, 2022-12-01:
Hi, I added better support for defrost functions on your model. I also found out that the fan speeds are available, so added those as well.

abstacek, 2022-12-01:
Hi, it works perfectly. Thank you. I have one more question. Do you think it is possible to somehow turn off the preheating of the air before the modbus? I noticed that the preheat takes over 1200W. It is a function of the polar units and it is possible to turn it off, but I have no idea how. It would be great if it was possible.
[image]

abstacek, 2022-12-01:
It would also be nice if the compressor could be turned off in a different way than by lowering the desired temperature. Ideally a switch in HA.

abstacek, 2022-12-01:
Here is the option in the HMI.
[image]
[image]
[image]

closed 2022-12-02 by veista