veista/nilan #7: Configuration guidance for Sharewave Ethernet-to-RS485 converter

MKBU2, 2022-12-06:
Hi,

My setup:
- Nilan EC9 / CTS602
- Waveshare RS485 TO ETH (B) converter https://www.waveshare.com/wiki/RS485_TO_ETH_(B)
- Raspberry Pi 3 + Home Assistant
- Nilan integration installed from HACS

[image]

[image]

I get "Failed to connect" when trying to add the integration. Any ideas where the problem would be:
1. Waveshare doesn't read Nilan
2. The integration doens't support EC9 although it's controlled by CTS602
3. IP-connectivity issue between HA and Waveshare 

The modbus integration used to work in Openhab and previous USB-converter.

veista, 2022-12-06:
Hi,

It propably is that your bridge is not ModBus TCP, but instead ModBus RTU over TCP. I havnt enabled official support for that yet, but you can try it by editing the device.py and replacing "tcp" with "rtuovertcp".

Nilan EC9 is not supperted yet, but I think you will help me with that.

veista, 2022-12-06:
If that does not help, then you will have to try modifying the settings on your bridge. Please share the settings if you get it to work, it may help in future issues.

veista, 2022-12-06:
Also I could add support for usb adapters, but I cannot debug this my self, so you would have to help me with that also and it might be difficult to do.

veista, 2022-12-06:
> > If that does not help, then you will have to try modifying the settings on your bridge. Please share the settings if you get it to work, it may help in future issues.
> 
> This was on my mind because Sharewave seems to be quite popular and your integration very easy to install.
> 
> The manual doesn't clearly state that the converter could act in Modbus TCP state (if it isn't the NONE mode)
> 
> [image]
> 
> Kiitos jo näistä vinkeistä.
> 

I would guess transparent is "rtuovertcp" and Modbus TCP is "tcp" so it should work. There might be some other setting that is wrong.

veista, 2022-12-06:
Baudrate should be 19200 i think. And also destination and source ports should be both 502.

veista, 2022-12-06:
Also make sure that your device id is 30. You can find this setting in the HMI menu.

MKBU2, 2022-12-06:
Changed the ports to 502.

With all custom components/nilan files as defaul I get:
Waveshare / Multi-host settings / Protocol = **none** 
-> Invalid response From the device

Waveshare / Multi-host settings / Protocol = **Modbus TCP to RTU**
-> Unsupported Device

Something is changing..

veista, 2022-12-06:
Hi, unsupported device is correct. Open a new issue with the information from the read-me instructions.

closed 2022-12-06 by veista