veista/nilan #8: Nilan EC9 Modbus - Incomplete message received

MKBU2, 2022-12-06:
As defined in https://github.com/veista/nilan/issues/7

the outcome in logger for pymodbus.transaction when trying to connect:

[pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.1.155', 60163)
[pymodbus.transaction] Current transaction state - IDLE
[pymodbus.transaction] Running transaction 1
[pymodbus.transaction] SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1e 0x3 0x3 0xe8 0x0 0x1
[pymodbus.client.sync] New Transaction state 'SENDING'
[pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
[pymodbus.transaction] Incomplete message received, Expected 64008 bytes Recieved 14 bytes !!!!
[pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
[pymodbus.transaction] RECV: 0xfa 0x7 0x13 0x2 0xfa 0x2 0x28 0x5f 0x2a 0x72 0xa5 0xc1 0xfa 0xff
[pymodbus.framer.socket_framer] Processing: 0xfa 0x7 0x13 0x2 0xfa 0x2 0x28 0x5f 0x2a 0x72 0xa5 0xc1 0xfa 0xff
[pymodbus.framer.socket_framer] Frame check failed, ignoring!!
[pymodbus.transaction] Getting transaction 1
[pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'

veista, 2022-12-06:
You need to enable debugging for custom_components.nilan

https://www.home-assistant.io/integrations/logger/

veista, 2022-12-06:
Yes. Then try setting the integration.

MKBU2, 2022-12-06:
This was the result in the logs when I tried to put up the integration.

veista, 2022-12-06:
For custom_components.nilan

MKBU2, 2022-12-06:
The error is "Invalid response From the device" with these logs.

MKBU2, 2022-12-06:
Integration tells : "Unsupported device"

and logs are:

(MainThread) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.1.155', 43239)
(MainThread) [pymodbus.transaction] Current transaction state - IDLE
(MainThread) [pymodbus.transaction] Running transaction 1
(MainThread) [pymodbus.transaction] SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1e 0x3 0x3 0xe8 0x0 0x1
(MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
(MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
(MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
(MainThread) [pymodbus.transaction] RECV: 0x0 0x2 0x0 0x0 0x0 0x5 0x1e 0x3 0x2 0x0 0x15
(MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x1e 0x3 0x2 0x0 0x15
(MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
(MainThread) [pymodbus.transaction] Adding transaction 1
(MainThread) [pymodbus.transaction] Getting transaction 1
(MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
(MainThread) [custom_components.nilan.config_flow] Device Type 21 not found in supported devices list

veista, 2022-12-06:
Nice. Now if you could provide the pictures.

veista, 2022-12-06:
I meant the pictures mentioned in the readme file. A picture of the device type plate and if you have a touch HMI a picture of the info page.

MKBU2, 2022-12-06:
[image]

veista, 2022-12-06:
Thanks, I will add a beta release ASAP, so I can get more information and add all of the entities your device supports.

MKBU2, 2022-12-06:
Seems to work! Excellent!

Modbus version = 8
Device Software version =2.31.d

Couple of errors showing in the logs:

2022-12-06 19:30:10.559 ERROR (SyncWorker_6) [homeassistant.components.modbus.modbus] Pymodbus: Nilan: Exception Response(132, 4, IllegalAddress)
2022-12-06 19:30:13.341 ERROR (SyncWorker_7) [homeassistant.components.modbus.modbus] Pymodbus: Nilan: Exception Response(131, 3, IllegalAddress)

veista, 2022-12-07:
Added better support with 1.0.22B. I couldn't find documentation for ModBus Version 8, so if some entities don't work or you get Modbus errors, please do tell.

MKBU2, 2022-12-09:
Seems to be working until the x.25b -> x.25 stopped the connection. I'll look at the logs.

My usermanual seems to be only in finnish.

Magifi, 2022-12-09:
Hi and thanks in advance for great integration. I'm quite noob with HA and now I'm building my home automation from scratch. At the moment I'm running Nilan with openHAB and USB-RS485 converter in my first raspberry pi. I assume that this has to be removed and buy i.e USR-TCP232-410S module to start configuring/building connection to your Nilan integration with my second raspberry pi (which will hopefully be the only device in future). I have old CTS602 (from 2015). Sorry for unrelated question.

veista, 2022-12-09:
> Seems to be working until the x.25b -> x.25 stopped the connection. I'll look at the logs.
> 
> My usermanual seems to be only in finnish.

Yes. I wont add your device  to master until it works in a sufficient manner. Use only the beta releases until then.

You can download the english version from https://www.en.nilan.dk/downloads/archives

veista, 2022-12-09:
> Hi and thanks in advance for great integration. I'm quite noob with HA and now I'm building my home automation from scratch. At the moment I'm running Nilan with openHAB and USB-RS485 converter in [...]

Please open a new issue if you have furter questions. At this point USB adaptors are not supported unfortunately. It will hopefully be added at some point in the future. So yes you assume correctly.

MKBU2, 2022-12-09:
This is Nilan EC9 models equivalent model VP18 M2 EK

[image]

Btw. for some reason HACS always dowloads 1.25 although you choose 1.25b or 1.23b beta version.

veista, 2022-12-09:
Yes, I know. I just want a picture from your physical HMI panel when you are at SHOW DATA -> TYPE. I am working on a bigger update to the integration and wanted to release tested work on 1.0.25. The reason why HACS keeps wanting to update to this is because it is newer. There will be a 1.1.0B available in the near future, which should in theory support your device properly.

MKBU2, 2022-12-09:
[image]
[image]
[image]

veista, 2022-12-12:
Did the invalid register errors go away with release 1.1.1?

closed 2022-12-14 by veista