veista/nilan #9: Mapping Device Version Numbers to Modbus Version Numbers

veista, 2022-12-08:
Hi,

I am trying to support all devices in the Nilan CTS602 family, but the documentation is kind of lacking. If you are using this Integration and have a device outside of the following list, please comment here with your device firmware version and ModBus Version as stated by the integration.

Thank You!

Firmware: ModBus Version
2.30.e: 7
2.31.d: 8
16.56.0: 22

Magifi, 2022-12-10:
Here’s mine:

[image]
[image]
[image]

veista, 2022-12-10:
Thanks, I would also need the Modbus Version. You can only find this info by reading it through ModBus.

Magifi, 2022-12-10:
> Thanks, I would also need the Modbus Version. You can only find this info by reading it through ModBus.

2.21.a:5

Found it from openHAB items.

llehti, 2022-12-19:
2.35.a: 9

AlvinKleine, 2022-12-21:
Firmware: 2.38.t
Modbus version 19

netforceatg, 2023-01-11:
VPL 25c / VPL 28c

Firmware: 2.37.c
Modbus version 11

sainio, 2023-03-12:
VP 18cek
Firmware: 1.6.26
ModBus Version: 21

Airler, 2023-09-22:
Firmware: 1.1.25
ModBus Version: 20

nmkloster, 2024-01-03:
Firmware: 2.1.34 - ModBus version: 21

ristoavila, 2024-03-22:
Firmware: 2.38r - Modbus Version: 19

veista, 2024-03-22:
Thank you all for helping out with this issue! I will be closing this since all of the open questions I had have been anwsered.

closed 2024-03-22 by veista