veista/nilan #10: Nothing works at all after the last update.

abstacek, 2022-12-09:
Nothing works at all after the last update. ( 1.0.25)
[image]

abstacek, 2022-12-09:
Updated to version 1.0.19

veista, 2022-12-09:
That is weird. Try removing and adding the integration again. Also could you report your ModBus version to the other Issue, thanks!

abstacek, 2022-12-09:
Where can I find the ModBus version?

veista, 2022-12-09:
> Where can I find the ModBus version?

It is a sensor in diagnostic. Scroll down on the device page.

abstacek, 2022-12-09:
Ok. Thanks. ModBus Version is 22.

veista, 2022-12-09:
Thank you! Did you get the 1.0.25 working. It should work normally on COMBI 302

abstacek, 2022-12-09:
Yes, I tried removing the integration and adding it again and it works.
I have two wishes that I would like to resolve if possible.

1. How to turn off the electric preheating of the air (polar version)?

2. Can the supply fan be stopped when defrosting the evaporator? I tried via node-red, but maybe I'm doing something wrong.

veista, 2022-12-09:
Hi, If these can't be done in the HMI, these can't be done by the Integration. The Integration can't add functionality to your device.

If these are possible in the HMI open  a new issue.

closed 2022-12-09 by veista