veista/nilan #11: Use component with USB-RS485 interface

szilardx, 2022-12-09:
Hi,
I would interested in trying out this component.

Currently I use a USB-RS485 stick with the Homeassistant modbus integration (https://github.com/StefanUG/nilan-hass-modbus/tree/master)

Would it be difficult to make it able to use the serial connection?
Thanks in advance!

veista, 2022-12-09:
At the moment the integration does not support usb adaptors, but it is possible to add. This is planned to be implemented in the future.

veista, 2022-12-11:
Just released initial version with serial support.

veista, 2022-12-12:
@magifi can you also test if this works for you?

Magifi, 2022-12-13:
> @Magifi can you also test if this works for you?

I already ordered RS485->ethernet device. I think I will try to run Nilan in HA and in openhab at the same time as long as I get my HA in order and stable. Perhaps I could use them both in modbus. Other one with RS485->ethernet adapter and other one with RS485 -> usb adapter.

But I will try my RS485-USB dongle as well with HA and then report the results.

closed 2022-12-14 by veista

llehti, 2022-12-21:
Waveshare USB to RS485 (https://www.waveshare.com/usb-to-rs485.htm) confirmed working.