veista/nilan #14: No Response received from the remote unit/Unable to decode response

gllmlbrt, 2022-12-13:
Hi, 
Currently on 1.1.1B, and since release 1.0.25 I have after many hours of perfect function and error as follow:
```
Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:391
Integration: Modbus (documentation, issues)
First occurred: 05:58:02 (1 occurrences)
Last logged: 05:58:02

Pymodbus: Nilan: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
```
And then all entities generate an error like this one:
```
Logger: custom_components.nilan.device
Source: custom_components/nilan/device.py:2204
Integration: Nilan (documentation, issues)
First occurred: 05:58:02 (988 occurrences)
Last logged: 08:42:34
Could not read get_time
```
Reloading the integration does not fix it.
Restarting Hass core does not fix it.
To make it work again I need to reboot the bridge and **then** reload the integration.
Then it work flowlessly for some 8-10 hours and crashes again...

I will try now to revert to pre-1.0.25 and see over long time if it occurs again...

My device is Comfort series we just added to the supported devices.
This did not occur before 1.0.25 ( in the beta versions we tested before).

veista, 2022-12-13:
That is weird. I have it running fine. It must be related to the bridge somehow.

veista, 2022-12-13:
P.s. there shouldn't be anything different in the newest version in your case.

gllmlbrt, 2022-12-16:
Issue appears to have disappeared in 1.1.2.B. maybe a coincidence... You can close the issue

veista, 2022-12-16:
Ok, it must be a coincidense.

closed 2022-12-16 by veista