veista/nilan #15: T4 sensor

abstacek, 2022-12-14:
Hi, I have a feeling if there is an error with the "Waste Air Temperature" display on the Combi 302 model. As the sensor.py file has the same name for the T4 and T6 sensors.
It is precisely the data from the T4 sensor that I am missing. You can look at it? Thank you.

[image]

veista, 2022-12-14:
Indeed. I will have to change the naming convention on these sensors since they are not consitent between models.

veista, 2022-12-14:
T6 is evaporator yes, but it is the last sensor before waste in your model too. Can you confirm that the waste temperature in you HMI is actually T4? On other models with a compressor it is T6.

veista, 2022-12-14:
And I dont mean the sensor name. I mean the temperature in the front panel picture.

abstacek, 2022-12-14:
The unit also has a passive heat exchanger. The latter triggers the defrost based on the T4 sensor. It also has a T6.
[image]
[image]
[image]

closed 2022-12-14 by veista

abstacek, 2022-12-15:
Hi. T2: Supply air sensor is now missing. On the contrary, I see T7 and the combi 302 does not have it. Maybe just misnamed.

veista, 2022-12-15:
T7 == T2 it is just the register name. I do not know how it internally works, but T7 always gives a value, T2 doesn't. This doesn't relate to the original issue and will not be fixed.