veista/nilan #19: CTS700 - what is needed?

Chrunchy, 2022-12-18:
Ive currently added my Air 9 with CTS700 trough modbus and the network cable attached to the system.
Currently mapping each adress (Mine is a none geo version).

What is needed to add support for CTS700 too? 

Mine is currently added to HA:
modbus:
  - name: nilan
    type: tcp
    host: 192.168.2.217
    port: 502
    sensors:
      - name: nilan_humidity
	# Current humidity
        address: 21776
        unit_of_measurement: "%"
        slave: 1
      - name: nilan_t20_outdoor_temperature
	# Outdoor unit temperatur
        unit_of_measurement: "°C"
        slave: 4
        address: 20688
        count: 1
        scale: 0.1
etc. etc.

Some data is placed at slave 1 (Indoor unit), and the outdoor unit data is placed on slave 4 for instance.
Latest ModBus guide i could find is dated back in 2018, but seems to be working somewhat atleast.
https://www.en.nilan.dk/Files/Files/Engelsk/Downloads/7.%20Modbus%20-%20BACnet/2018_04_Modbus_CTS700_Modbus_protokol.pdf

veista, 2022-12-18:
Hi,

It's great that you are interested in this project! There are a lot of changes needed to integration to accommodate cts700. If you are willing to help me with this endeavor I think we can succeed. I have never even seen a cts700 device so you will have to help me with testing, and give me detailed explinations with pictures etc. If needed. Remember that the integration should support all cts700 devices, not just your model/version.

I will start by writing up the registers. After that we will try to figure out how to saftely recognise your device through the integration.

When this is all done we can start adding functionality.

veista, 2022-12-20:
Please respond if you are up for this? I can not do this by myself.

Chrunchy, 2022-12-21:
I'm definitely up for helping with what i can, but i will have limited time from christmas and untill mid january. 
Currently i'm importing all registers from nilans documents for the CTS700. Of course i can't test everything as i don't have the geo model as mine is the Air9 with a air to water heatpump for centralheating. It's a limited amount of different registers for the GEO model.

The latest documentation i found for the CTS700 is on Nilan's swedish site with a cert issue since july, https://www.nilan.se/Admin/Public/DWSDownload.aspx?File=%2fFiles%2fFiler%2fDownload%2fDanish%2fDokumentation%2fSoftware+vejledninger%2fModbus%2fCTS700_Modbus_protokol.pdf .

I know there is an even older document with some other registers for the same sensors, however some of those stopped working on mine after i had it sw updated by Nilan. These atleast work, and the document is dated to 2018.

jakobmoll, 2022-12-22:
Hi

I'm currently using a manual modbus over tcp configuration for my Nilan Compact P (CTS700). The package I've setup for my integrations is listed here - perhaps this could help?

https://gist.github.com/jakobmoll/96d67a6e3c98b1742230057c5923da63

veista, 2023-01-10:
Hi,

I'm in a bit of a loss here.

So I guess the older versions of Compact P use the manual found here https://symlink.dk/stuff/CTS700_MODBUS-rev%202.01.pdf

And the newer versions with Air and Geo support use the manual found in @Chrunchy link.

I'm not sure how to support both, I cannot seem to find a version register on the newer document.

Can you @Chrunchy  test register 4632 if you can read the version number?

mbobinac, 2023-06-13:
Hi,

i have Nilan CTS700 and looking for integration support. My version is 1.40.05.38 from 2015 and has not been updated since.
I have a client app i can login as Installer (uses MODBUS to connect to CTS700) version 1.4.139 from 1.4.2018.

I can connect to CTS700 via the app from my server, but if i use MODBUS tools (Modbus pool) it says read error when connecting to CTS700. Not sure if permission or something else, i need password to login as Installer, but should not need anything to go as User. The Nilan App just does not allow anything else as Installer.

I've installed your HACS code, but when connecting to Nilan says "Invalid Response from Device" which is expected as you probably check some version data in one of the registers.

Cheers

convicte, 2023-09-20:
Hi @veista,

I've ended up purchasing an Elfin EW11A unit to integrate my Compact S pump, but struggle with finding both documentation (modbus registrars for compact S) as well as ready integrations.

If you could advise on what you'd need, I would be happy to at least try.

veista, 2023-09-20:
> Hi @veista,
> 
> I've ended up purchasing an Elfin EW11A unit to integrate my Compact S pump, but struggle with finding both documentation (modbus registrars for compact S) as well as ready integrations.
> 
> If you could advise on what you'd need, I would be happy to at least try.

I believe Compact S is a cts602 device so your question is in the wrong place. You can open a discussion, if you need generic help. I can not help with your setup: Read previous issues for hints, follow the instructions on the front page and then we can add Compact S to the supported devices.

lauer, 2023-10-08:
Hi @veista 
I am also using @jakobmoll 's manuel modbud sensor config to interact with my Nilan Compact P Air 9 - CTS700. 

Even though python is not my main language, I might be able to help testing and fixing stuff. 
Right now, the plugin gives the following error when trying to connect

```
  File "/config/custom_components/nilan/config_flow.py", line 108, in async_step_tcp
    await async_validate_device(
  File "/config/custom_components/nilan/config_flow.py", line 67, in async_validate_device
    result.registers[0].to_bytes(2, "little", signed=False),
    ~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

But it could be a UnitID problem, because I can't see why it should be 30 as default. If its the same as slave, it should have been '1', but that doesn't work either.

veista, 2023-10-08:
Hi,

Cts700 works differently than cts602. I have some modifications to support cts700 in my code, but at the moment absolutely NO functionality has been implemented. I have slowly made it even theoretically possible to support cts700. Adding support will take time and I am a busy man.

Thanks for your understanding, I will report here once there is something to test.

lauer, 2023-10-08:
okay, so its not just creating a registry mapping for cts700 like the one for 602? 
Could you create a branch with you work until now?

veista, 2023-10-08:
There is no branch yet. There is no code yet. It is complicated to add since I have not originally thought about adding cts700.

Keep in mind that there is also two types of cts700 as I mentioned before in this issue. Both are completely different.

Johannes-288, 2023-10-18:
Hello everyone, this is my configuration for my CTS 700. Unfortunately I encountered a problem: Switches are recognized as errors. Can anyone help me with that?

modbus:
  - name: nilan
    type: tcp
    host: 192.168.88.8
    port: 502

    close_comm_on_error: true
    delay: 0
    message_wait_milliseconds: 30
    retries: 3
    retry_on_empty: false
    timeout: 5

    sensors:
      - name: nilan_t0_humidity
        unique_id: nilan_t0_humidity
        address: 21776
        unit_of_measurement: "%"
        slave: 1

      - name: nilan_t1_outdoor
        unique_id: nilan_t1_outdoor
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20282
        scale: 0.1
        count: 1
        precision: 2

    switch:
      - name: nilan_compact_p_set_ventilation_on_pause
        unique_id: set_ventilation_on_pause
        slave: 1
        address: 20100
        count: 1
        write_type: coil

maisun, 2024-01-10:
Hi, I have tried to setup CTS700 myself with OH Modbus integration, I’d like to ask if anybody knows which register to read containing the below info:
1) Compressor state: can be off, defrosting, hot water production, heating, cooling
2) Flag indicate if the supplementary eletrical heater for the water tank is running or not (the red dot on the display)
Many thanks!

lauer, 2024-02-04:
I know you havn't written that it was ready. 

However, I would like to give the new 1.12.13b on my Nilan CompactP Air a try, but I just get "invalid response" with no further log in console, when I try to add the integration. 

I fetch data using modbus plugin in HA using the same IP, port and device (1), so I know it answers, but could it give problems, what some processes are already communicating with it?

Please tell me, how I can help testing it?

veista, 2024-02-04:
Thanks for your enthusiasm. I will let you know when I need your help. It will take some time to get to that point.

Do you have the old or new cts700?

leonkdk, 2024-02-05:
I have a CTS700 Touch.
Display Software Version: 09.06.45 (2017-07-08)
Print Software Version: 2.02.13.93 (2017-07-08)
I am very interested in a working connection to my Home Assistant and as such would like to offer my help as well!

Johannes-288, 2024-02-05:
Hello everyone, I read out my Nilan CompctP GEO 6 via Modbus and also controlled some registers. Unfortunately, the YAML is not completely finished yet.

modbus:
  - name: nilan
    type: tcp
    host: >your IP-address<
    port: 502

    delay: 0
    message_wait_milliseconds: 30
    retries: 3
    timeout: 5

    sensors:
      - name: nilan_t0_feuchte
        unique_id: nilan_t0_feuchte
        address: 21776
        unit_of_measurement: "%"
        slave: 1
      - name: nilan_t1_aussentemperatur
        unique_id: nilan_t1_aussentemperatur
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20282
        scale: 0.1
        precision: 1
      - name: nilan_t2_zulufttemperatur
        unique_id: nilan_t2_zulufttemperatur
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20284
        scale: 0.1
        precision: 1
      - name: nilan_t3_raumablufttemperatur
        unique_id: nilan_t3_raumablufttemperatur
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20286
        scale: 0.1
        precision: 1
      - name: nilan_t4_temperatur_nach_warmetauscher
        unique_id: nilan_t4_temperatur_nach_warmetauscher
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20288
        scale: 0.1
        precision: 1
      - name: nilan_t5_temperatur_nach_waermepumpe
        unique_id: nilan_t5_temperatur_nach_waermepumpe
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20290
        scale: 0.1
        precision: 1
      - name: nilan_t6_temperatur_waermepumpenverdampfer
        unique_id: nilan_t6_temperatur_waermepumpenverdampfer
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20292
        scale: 0.1
        precision: 1
      - name: nilan_t7_temperatureingang_nach_heizung
        unique_id: nilan_t7_temperatureingang_nach_heizung
        device_class: temperature
[... 373 lines cut ...]
        name: Nilan Systemalarm
        device_class: problem
        state: >-
          {% if states('sensor.nilan_alarm') |float(0) == 1 %}
          Ok
          {% else %}
          Fehler
          {% endif %}
      - unique_id: Nilan Lüftungsalarm
        name: Nilan Lüftungsalarm
        device_class: problem
        state: >-
          {% if states('sensor.nilan_compact_p_alarm') |float(0) == 1 %}
          Ok
          {% else %}
          Fehler
          {% endif %}
      - unique_id: Nilan GEO Zusatzheizung
        name: Nilan GEO Zusatzheizung
        state: >-
          {% if states('sensor.nilan_compact_p_alarm') |float(0) == 1 %}
          Ein
          {% else %}
          Aus
          {% endif %}

veista, 2024-04-06:
I have finally started to create development tools for implementing cts700 devices. 

It would be great help if owners of cts700 devices would dump all registers to a file with qmodmaster or equivalent accompanied with device information and a picture of the type plate.

Johannes-288, 2024-04-07:
Hello everyone, here is the yaml in a package of my Nilan Compct P Geo 6, which has already been working for a few weeks! Unfortunately, a few little things are still missing (for example: manual control of fan speed 1-4)
A small note: I have translated my yaml from German into English. There might have been mistakes

modbus:
  - name: nilan
    type: tcp
    host: >xx.xx.xx.xx<
    port: 502

    delay: 0
    message_wait_milliseconds: 30
    retries: 3
    timeout: 5

    sensors:
      - name: nilan_t0_Actual_air_humidity
        unique_id: nilan_t0_actual_air_humidity
        address: 21776
        unit_of_measurement: "%"
        slave: 1
      - name: nilan_t1_outdoor_air_temperature
        unique_id: nilan_t1_outdoor_air_temperature
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20282
        scale: 0.1
        precision: 1
      - name: nilan_t2_supply_air_temperature
        unique_id: nilan_t2_supply_air_temperature
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20284
        scale: 0.1
        precision: 1
      - name: nilan_t3_extract_air,_room_temperature
        unique_id: nilan_t3_extract_air,_room_temperature
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20286
        scale: 0.1
        precision: 1
      - name: nilan_t4_discharge_air,_after_heat-discharge_exchanger
        unique_id: nilan_t4_discharge_air,_after_heat-discharge_exchanger
        unit_of_measurement: "°C"
        slave: 1
        address: 20288
        scale: 0.1
        precision: 1
      - name: nilan_t5_discharge_air,_after_heat_pump
        unique_id: nilan_t5_discharge_air,_after_heat_pump
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20290
        scale: 0.1
        precision: 1
      - name: nilan_t6_evaporator_temperature
        unique_id: nilan_t6_evaporator_temperature
        device_class: temperature
        unit_of_measurement: "°C"
        slave: 1
        address: 20292
        scale: 0.1
        precision: 1
      - name: nilan_t7_supply_air_temperature_afterheater
        unique_id: nilan_t7_supply_air_temperature_afterheater
        device_class: temperature
[... 346 lines cut ...]
        device_class: problem
        state: >-
          {% if states('sensor.nilan_geo_brine_pressure') |float(0) == 1 %}
          Ok
          {% else %}
          Error
          {% endif %}
      - unique_id: Nilan Geo Alarm
        name: Nilan Geo Alarm
        device_class: problem
        state: >-
          {% if states('sensor.nilan_geo_alarm') |float(0) == 1 %}
          Ok
          {% else %}
          Error
          {% endif %}
      - unique_id: Nilan Compact P Alarm
        name: Nilan Compact P Alarm
        device_class: problem
        state: >-
          {% if states('sensor.nilan_compact_p_alarm') |float(0) == 1 %}
          Ok
          {% else %}
          Error
          {% endif %}

terkilddk, 2024-04-25:
I have a CTS700 device and would like to help, i use this config for my modbus connection today.
https://pastebin.com/PZNHBRXM

I will be happy to help with qmodmaster if someone can guide me.

lauer, 2024-04-26:
Same here, however, I only have macs and linux servers, and most of the tools seems to require windows :/

maisun, 2024-04-26:
> I have a CTS700 and a Nilan compact P Air9 and would like to help, i use this config for my modbus connection today. https://pastebin.com/PZNHBRXM
> 
> Software is updated to latest version by Nilan last week
> 
> I will be happy to help with qmodmaster if someone can guide me.

Thanks for sharing your config! Have you figured out how to control active cooling? Neither 3284 nor 5288 works for me, already tried to Google for years with no clue....

veista, 2024-04-26:
> Same here, however, I only have macs and linux servers, and most of the tools seems to require windows :/

What do you mean? For example qmodmaster works on linux and is open source. You just need to compile it yourself.

veista, 2024-04-26:
> I have a CTS700 and a Nilan compact P Air9  and would like to help, i use this config for my modbus connection today.
> https://pastebin.com/PZNHBRXM
> 
> Software is updated to latest version by Nilan last week
> 
> I will be happy to help with qmodmaster if someone can guide me.
> 

What kind of guidance do you need? The software is pretty straight forward.

veista, 2024-04-26:
What I need is: 1. dump of all register contents. This can be tedious. 2. A picture of the device type plate. 3. A clear correlation between register in modbus and function in HMI

maisun, 2024-06-01:
> What I need is: 1. dump of all register contents. This can be tedious. 2. A picture of the device type plate. 3. A clear correlation between register in modbus and function in HMI

Hi @veista : today I tried the tool QModMaster however couldn't really get it to work. I have the "older" version of CTS700 so all registers are below 10000. How can I dump all registers with their contents? It can only read max 100 at a time, and looks to me like if there is any unknown register it will just fail to get the rest values. 
I think I know the addresses of almost everything except for Active Cooling - the address is not documented in the ModBus technical document, so the best approach is to find a tool that can somehow can dump all registers and compare before/after the change.

veista, 2024-06-01:
> Hi @veista : today I tried the tool QModMaster however couldn't really get it to work. I have the "older" version of CTS700 so all registers are below 10000. How can I dump all registers with their [...]

As I said this can be tedious. You can write your own dump tool that suits your needs better easily with python.

maisun, 2024-06-01:
> As I said this can be tedious. You can write your own dump tool that suits your needs better easily with python.

Ok thanks! I don’t think I have the skills and time for that sadly.

jellevictoor, 2024-07-21:
I have everything i could find mapped out for the cts700 for the compact P + geo 3.
when reading this i see a correlation between the Air and the geo.
the second board is apparently on the modbus ID 4, while the compact p's mainboard is on modbus ID 1.

I'll create a PR with the registers

maisun, 2024-07-21:
> I have everything i could find mapped out for the cts700 for the compact P + geo 3. when reading this i see a correlation between the Air and the geo. the second board is apparently on the modbus ID 4, while the compact p's mainboard is on modbus ID 1.
> 
> I'll create a PR with the registers

Thanks, have you managed to find what register is for controlling active cooling? It's not mentioned in any document I have found from Nilan. Thanks

jellevictoor, 2024-07-21:
> Thanks, have you managed to find what register is for controlling active cooling? It's not mentioned in any document I have found from Nilan. Thanks

i dont think i have that one exactly.
i have 2 documents, one with wrong addresses and no explanation, and one with the right adresses and the correct explanation.
[WRONG_ADDRESS cts700 description.pdf](https://github.com/user-attachments/files/16325404/WRONG_ADDRESS.cts700.description.pdf)
[Modbus_Registers_Description_CTS700.pdf](https://github.com/user-attachments/files/16325410/Modbus_Registers_Description_CTS700.pdf)

maisun, 2024-07-21:
> i dont think i have that one exactly. i have 2 documents, one with wrong addresses and no explanation, and one with the right adresses and the correct explanation. [WRONG_ADDRESS cts700 [...]

Thanks for the reply! I have searched for a couple of years and couldn't find out what register they use to control active cooling. Also tried to contact their technical support without luck. My idea is to control active cooling based on temperature and electricity price etc.

jellevictoor, 2024-07-21:
i have found multiple documents over the years, over when I even still used openhab. i think i posted everything there then.
now i have a lot of info, but my motherboard seems to be losing connection to the controller once and a while. to resolve this i need to physically remove the power cable from the mainboard.
i dont really know what is causing this.

FYI, here is my home assistant config atm for a compact p (xl don;t know if that matters) + geo 3 module
the ones in comment work, but i didnt need them

```- name: nilan
  type: tcp
  host: 192.168.1.11
  port: 502
  message_wait_milliseconds: 30
  binary_sensors:
    # - name: Nilan_do1_compressor
    #   address: 5284
    #   input_type: holding
    #   slave: 1
    - name: Nilan_do2_dwh_heater
      address: 5285
      input_type: holding
      slave: 1
    - name: Nilan_dwh_heater_enabled
      address: 3938
      input_type: holding
      slave: 1
    # - name: Nilan_do3_user_select_to_output
    #   address: 5286
    #   input_type: holding
    #   slave: 1
    # - name: Nilan_do4_bypass_damper
    #   address: 5287
    #   input_type: holding
    #   slave: 1
    - name: Nilan_do5_air_heating_valve
      address: 5288
      input_type: holding
      slave: 1
    - name: Nilan_do6_dhw_heating_valve
      address: 5289
      input_type: holding
      slave: 1    
    # - name: Nilan_do7_dhw_preheater_relay
    #   address: 5290
    #   input_type: holding
    #   slave: 1    
    # - name: Nilan_do8_dhw_allow_external_heating
    #   address: 5291
    #   input_type: holding
    #   slave: 1
    # - name: Nilan_di1_high_pressure_switch
    #   address: 5096
    #   input_type: holding
    #   slave: 1
    # - name: Nilan_di2_freeze_overheat
    #   address: 5097
    #   input_type: holding
    #   slave: 1
    # - name: Nilan_di3_user_select
    #   address: 5098
    #   input_type: holding
    #   slave: 1
    # - name: Nilan_di4_deicing_thermostat
    #   address: 5099
    #   input_type: holding
    #   slave: 1
    # - name: Nilan_di5_fire_thermostat
    #   address: 5100
    #   input_type: holding
    #   slave: 1  
    # - name: Nilan_di6_user_select_2
[... 399 lines cut ...]
          {% if states('sensor.Nilan_param_pause_mode')| int == 0 %} on {% else %} off {% endif %}
        percentage_template: "{{ (states('sensor.nilan_param_userfanspeed') | float) * 25 }}"
        turn_on:
          service: modbus.write_register
          data:
            hub: nilan
            slave: 1
            address: 4727
            value: 0
        turn_off:
          service: modbus.write_register
          data:
            hub: nilan
            slave: 1
            address: 4727
            value: 1
        speed_count: 4
        set_percentage:
          service: modbus.write_register
          data_template:
            hub: nilan
            slave: 1
            address: 4747
            value: "{{ (percentage/25) + 100 }}"
        speed_count: 4```

MarcelPrinsen, 2024-08-01:
This modbus protocol also works (file from 2018):
[2018_04_Modbus_CTS700_Modbus_protokol.pdf](https://github.com/user-attachments/files/16457868/2018_04_Modbus_CTS700_Modbus_protokol.pdf)

maku1958, 2024-12-13:
I tried a lot of the inputs posted here. But it is with all register-IDs the same. Entity not found. 
Here is my yaml:
`  - name: "NCS700"
    type: rtuovertcp
    host: 192.168.188.107
    port: 502
    # username: admin
    # password: 6699
    message_wait_milliseconds: 300
    timeout: 5
    delay: 5
    binary_sensors:
       - name: Next-Out-Filter-Change
         address: 1329
         slave: 1
       - name: Next-In-Filter-Change
         address: 1328
         slave: 1
    climates:         
       - name: Nilan_Temp_T3_Extract
         unique_id: T3_Extract_Air
         address: 5154
         slave: 1
         data_type: int16
         scale: 0.1
         precision: 2
         target_temp_register: 5154
         input_type: holding 
         temperature_unit: "°C"
    sensors:
       - name: Average Humidity
         unique_id: Average_Value_Humidity
         address: 21776
         slave: 1
         # scale: 2
         # target_temp_register: 21776
         data_type: int32
         input_type: holding
         unit_of_measurement: '%'
       - name: Register PLC
         address: 20003
         slave: 1
         data_type: int16
         input_type: holding`

MarcelPrinsen, 2024-12-31:
And when you try this in your YAML:

# NILAN CTS700 modbus integratie
modbus:
  - name: nilan
    type: tcp
    host: 192.168..188.107
    port: 502
    climates:
      - name: Kamer Temperatuur
        slave: 1
        target_temp_register: 4746
        address: 5154
        min_temp: 10
        max_temp: 30
        temp_step: 0.5
        # data_type: int
        scale: 0.1
        precision: 1
      - name: Vloer Temperatuur
        slave: 4
        target_temp_register: 20680
        address: 5169
        min_temp: 10
        max_temp: 45
        temp_step: 1
        # data_type: int
        scale: 0.1
        precision: 1
      - name: Boiler Temperatuur
        slave: 1
        target_temp_register: 20460
        address: 5162
        min_temp: 40
        max_temp: 60
        temp_step: 1
        # data_type: int
        scale: 0.1
        # precision: 1
    switches:
      - name: nilan_allow_active_cooling
        slave: 1
        address: 20180
        command_on: 1
        command_off: 0
      - name: pause_dhw_production
        slave: 1
        address: 20440
        command_on: 1
        command_off: 0
        scan_interval: 1
      - name: nilan_fan_speed_2
        slave: 1
        address: 4747
        command_on: 102
        command_off: 101
      - name: nilan_fan_speed_3
        slave: 1
        address: 4747
        command_on: 103
        command_off: 101
      #        scan_interval: 1
      - name: pause_central_heating
        slave: 4
        address: 20600
        command_on: 1
        command_off: 0
        scan_interval: 1
      - name: nilan_air_cooling
        slave: 4
[... 360 lines cut ...]
        # 4709 = Compressor4
      - name: nilan_water_compressor_4
        #unit_of_measurement: 'days'
        slave: 1
        address: 4709
        data_type: int16
        # 4710 = Compressor5
      - name: nilan_water_compressor_5
        #unit_of_measurement: 'days'
        slave: 1
        address: 4710
        data_type: int16
        # 4711 = Compressor6
      - name: nilan_water_compressor_6
        #unit_of_measurement: 'days'
        slave: 1
        address: 4711
        data_type: int16
      - name: nilan_t20_outdoor
        unit_of_measurement: "°C"
        slave: 4
        address: 5171
        scale: 0.1
        precision: 1
        data_type: int16

MarcelPrinsen, 2024-12-31:
And when you try this YAML  
in your confidhuration YAML (assuming the IP 192.168.188.107 is within your LAN range):

[NILAN CTS700 modbus integratie YAML.docx](https://github.com/user-attachments/files/18282285/NILAN.CTS700.modbus.integratie.YAML.docx)

NILAN CTS700 modbus integratie YAML.docs

maku1958, 2024-12-31:
I'll give it a try.

MarcelPrinsen, 2025-01-01:
There are indeed adresses 4756 (inlet) and 4757 (outlet). Not in the YAML above. Toggle to 1 = reset. Although exchanging the filters is something you do at the unit itself with the unit turned off. For me this reset will be done at the unit itself. Also to prevent an accidentally reset with HA.

maku1958, 2025-01-01:
Good advice.
I did it the last years directly on the Nilan. But my Nilan and the CTS700touch are not in the same Room anymore.
I think about it, to put the Toggle on an extra page, with admin rights only, so there's only me to blame.

master3395, 2025-02-28:
> Hi
> 
> I'm currently using a manual modbus over tcp configuration for my Nilan Compact P (CTS700). The package I've setup for my integrations is listed here - perhaps this could help?
> 
> https://gist.github.com/jakobmoll/96d67a6e3c98b1742230057c5923da63

I recently purchased a USR-W630, which I plan to use with my Nilan Compact P (CTS700) and integrate with Home Assistant for system control.

However, I'm currently stuck on one key issue—how to properly connect it. Should I use the WAN/LAN port, or the LAN port with the orange RJ45 connector from the Nilan MB? Or is the RS485 terminal the only viable option?

From what I can tell, the top-left wires seem to be leading to the panel. The panel itself is located in another room, with the cable exiting the box, passing through a dedicated white box on top of the Nilan unit, and then continuing through the walls.

Any guidance on the correct connection method would be greatly appreciated!

Attached are images for reference:

[image]

[image]

[image]

MarcelPrinsen, 2025-03-01:
When you haev the CTS700 system you just connect the red LAN kable to your Local Network or Router. I tried a PUSR USR-W610 to connect the Nilan with the red LAN cable to the wireless network (STA mode I beleive), but I ran into issues with connection stabilty over time. Tried several times but couldn't get it stable. I now use a Netgear EAX15 and this has been rock solid ever since. 

[image]
[image]

master3395, 2025-03-01:
@MarcelPrinsen But if the red RJ45 cable is too short, is it a normal wire, or do I have to re-write a normal Cat5e or Cat6 cable?
I do have the USR-W630 I could have a try, since it's easier to wire a power cable around the door, than to re-wire a network cable.

MarcelPrinsen, 2025-03-01:
> [@MarcelPrinsen](https://github.com/MarcelPrinsen) But if the red RJ45 cable is too short, is it a normal wire, or do I have to re-write a normal Cat5e or Cat6 cable? I do have the USR-W630 I could have a try, since it's easier to wire a power cable around the door, than to re-wire a network cable.

The red one is a standard LAN RJ45 cable (Cat5e or Cat6 doesn't matter in this case) which you can extend with a LAN connector, and it needs only to be connected to your Router. No RS485 interface needed. Sorry for my ignorance if I don't understand your question. Bottom line is you need to connect the Nilan to your LAN and reserve a fixed IP address in your DHCP.

master3395, 2025-03-01:
I did try to connect it with an Cat8 cable now, since the cable that is in there doesn't seem to be long enough to go to the router.
But i don't get it showing on the router at all.

I also tried setting the W630 as an AP but with Modbus over IP, but as an AP it does not connect to my Network at all it seems.

[image]

[image]

MarcelPrinsen, 2025-03-01:
You must first change the IP adress in Nilan to match your range. I have 192.168.10.xxx (see below). From what I recall, 192.168.05.107 is the innitial address from Nilan. Change it to the desired adress in the Nilan and when you have established a connectiont, then you reserve thea address in your router to make sure no other client will take the IP address in the future.

[image]

MarcelPrinsen, 2025-03-01:
Yes you may: Discord-> marcel_33630

MarcelPrinsen, 2025-03-01:
BTW: the IP adress from your Nilan has to be set in the YAML. 

[image]

See below.

maisun, 2025-03-25:
Hi, I have Nilan Compact P Geo 3 with CTS700
There are 2 slaves, 1 for vent/hot water, 2 for geothermal heating.
I have scanned through all registers and currently made everything to work in OpenHAB. If it could help you I can send you my OH configuration (it's very different from HA but very self-explanatory).

master3395, 2025-03-27:
> Hi, I have Nilan Compact P Geo 3 with CTS700 There are 2 slaves, 1 for vent/hot water, 2 for geothermal heating. I have scanned through all registers and currently made everything to work in [...]

If you want, that would be nice.
I do have a file from [MarcelPrinsen](https://github.com/MarcelPrinsen), and i've asked ChatGPT to use all from the PDF that followed my system.
But atm my ISP's router, doesn't hold static IP's, so i will have to move one of my better routers to it's place.

maisun, 2025-03-27:
> If you want, that would be nice.

Here you go. Explanation, find items such as below:
`Number:Temperature      VAL_DEV_TSENS2
                        "Ventilation Supply Air Temperature Inlet [%.1f %unit%]"
                        <temperature>
                        (gNilanVentilation)
                        {channel="modbus:data:Ventilation:TemperatureRead:input5153:number"[profile="modbus:gainOffset", gain="0.1 °C", pre-gain-offset="0"], unit="°C"}
`

- VAL_DEV_TSENS2: should match the document name
-  "Ventilation Supply Air Temperature Inlet [%.1f %unit%]": description which is self explanatory 
- channel="modbus:data:Ventilation:TemperatureRead:input5153:number" register 5153 with input type (coil I think but in OH doesn't seem to matter)
- [profile="modbus:gainOffset", gain="0.1 °C", pre-gain-offset="0"]: basically the number needs to be divided by 10

There are some transformation needed for temperature and fan speed (101-104 I believe). I have been using this for years and it works correctly and as you can see I created Homekit heater/cooler based on it. 

Question: I'm new to HA, I read somewhere that HA Homekit bridge doesn't support Heater/Cooler, is it true? In the document I find Climate maps to Thermostat, no mentioning of heater/cooler...

[Nilan.txt](https://github.com/user-attachments/files/19493274/Nilan.txt)

master3395, 2025-03-28:
> Here you go. Explanation, find items such as below: `Number:Temperature VAL_DEV_TSENS2 "Ventilation Supply Air Temperature Inlet [%.1f %unit%]" <temperature> (gNilanVentilation) [...]

I used ChatGPT to merge all the information from my PDF's.

[2018_04_Modbus_CTS700_Modbus_protokol.pdf](https://github.com/user-attachments/files/19512296/2018_04_Modbus_CTS700_Modbus_protokol.pdf)

[EL Skjema CTS 700 September 15.pdf](https://github.com/user-attachments/files/19512298/EL.Skjema.CTS.700.September.15.pdf)

Here's the HA config file that ChatGPT made, from ALL config options.
My Nilan CTS700, uses the normal RJ45 cable, to a static IP.
I will do more documentation, once i get a working router with static IP.

[configuration.txt](https://github.com/user-attachments/files/19512535/configuration.txt)

maisun, 2025-03-28:
> Here's the HA config file that ChatGPT made, from ALL config options.
> My Nilan CTS700, uses the normal RJ45 cable, to a static IP.
> I will do more documentation, once i get a working router with static IP.

Please be aware that there are two major version of CTS-700, the pdf you attached is the newer version.
Most CTS700 on old firmware are using the registers in the document attached - it's not complete and I had to use ModBus scanners to find all fields.

[CTS700-ModbusRegisters-20170726.pdf](https://github.com/user-attachments/files/19512591/CTS700-ModbusRegisters-20170726.pdf)

master3395, 2025-03-28:
My PDF followed the home, when i moved in, it was in the full spec USB.
So my file should be correct for my system.

master3395, 2025-03-28:
Not sure how i want the dashboard.
But even if I don't have the connection up and running yet.
I got ChatGPT to make me a dashboard.

[image]
[image]
[image]
[image]
[image]