veista/nilan #21: VP 18cek: Additional holding registers

MKBU2, 2022-12-18:
Would it be possible to add holding registers:

Output.CenCircPump | 118 | EK circulation pump | VGU-VP
Output.CenHeat_1 | 119 | EK heater relays | VGU-VP
Output.CenHeat_2 | 120 | -- | VGU-VP
Output.CenHeat_3 | 121 | -- | VGU-VP

It would show user which central floor heating relays are on and if the circulating pump is on.

Also I think the VP 18CEK should also be able to show:
Input.T13_Return | 213 | 100 | °C | EK return temperature | VGU-VP

It gives user feedback how much heat has been lost during the circulation

veista, 2022-12-18:
Yes it would. Though just to save time and effort, you could check if these registers actually have values. You can use qmodmaster to do this. https://sourceforge.net/projects/qmodmaster/

MKBU2, 2022-12-18:
The pump and relays were shown in my old openhab configuration. It helped to find broken 1st relay issue.

Cant remember  if the return temperature was there but I would guess the system would use that value to adjust heating need. I'll check if I can see it.

closed 2022-12-20 by veista

MKBU2, 2022-12-22:
Does closing the issue mean that feature would come in next update or something else?

veista, 2022-12-22:
Yes. The holding registers you requested should be in 1.1.4B. Feel free to re-open the issue if not so.