veista/nilan #22: Unsupported Device: VP 18ek

llehti, 2022-12-19:
Trying to install the integration raises error: Unsupported Device.
Debug log reads:  
`2022-12-19 18:35:13.263 DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 20 not found in supported devices list`

QModMaster reveals the following:
Bus.Version: 9
App.VersionMajor: 2.
App.VersionMinor: 35
App.VersionRelease: a

On CTS 602 HMI:
TYPE: VP 18ek
SOFTWARE: 1 2.35.a

The device should essentially be the same as VP 18cek, but missing the cooling feature.

veista, 2022-12-19:
Hi, I will add the device ASAP.

llehti, 2022-12-20:
Managed to get it working by myself with trivial additions to device_map (ie. adding the entry for device 20: "VP 18ek" to CTS602_DEVICE_TYPES and the appropriate entries to CTS602_ENTITY_MAP). 
Even though the machine has some of the same entities as device 21 with bus version 22 accessible through HMI (central heating curves for example), the integration returns only unknown values. Possible address mismatches between bus versions? Will investigate further later, right now interested in actually using the integration.
Thank you for the great work!

veista, 2022-12-20:
You cannot access those (bus version 22 registers) with a bus version 9. You will have to get an update from Nilan for your device. I will go through the correct registers for your device, propably today.

veista, 2022-12-20:
Could you take a picture of your type plate? I cannot find the correct manual. Thanks.

llehti, 2022-12-20:
[image]

veista, 2022-12-20:
That is funny. It might be that the cooling function is disabled only by software. I wonder what would happen if you tried to write 21 to holding register 1000. Would you get the cooling function?

llehti, 2022-12-20:
I will try in the new year.

veista, 2022-12-20:
I guess the only register that is different is the cooling setpoint, though the manual does not mention any differences.

llehti, 2022-12-20:
Right now, the mode can be set to cooling on the front page of the HMI, but all it does is stop the compressor. There is no cooling menu entry visible in my HMI to set the cooling setpoint or boosted fan etc.

veista, 2022-12-20:
> Right now, the mode can be set to cooling on the front page of the HMI, but all it does is stop the compressor. There is no cooling menu entry visible in my HMI to set the cooling setpoint or boosted fan etc.

Ok, good to know, will remove the cooling ventilation step also.

closed 2022-12-20 by veista

veista, 2023-01-06:
> I will try in the new year.

Did you try this?

llehti, 2023-01-31:
> > I will try in the new year.
> 
> Did you try this?

Did not try yet, I wanted to confirm first that the device actually has the 4-way valve physically present, which is necessary for the cooling to work in the first place.
And it actually has it!
I will resume the modbus side experiment once the heating period is over, I don't want to risk bricking the controller in the middle of winter.
But seems very promising, never the less.

llehti, 2023-07-30:
@veista 
I finally had time to try it out, and I can confirm that it works!
Wrote 21 to holding register 1000, and immediately all VP18cek features appeared. 
Everything is fully functional in HMI and through the integration, and confirmed that the cooling feature working correctly.

veista, 2023-07-30:
That is awesome! Wonder what is the pricing difference originally. I don't think they sell the models without cooling anymore.