veista/nilan #23: Nilan <-> Waveshare RS485-ethernet <-> HA Nilan integration 'failed to connect' error.

Magifi, 2022-12-19:
Hi again and sorry for bothering with stupid questions (perhaps)

My setup:
Nilan EC CTS602
Waveshare RS485 TO ETH (B) converter https://www.waveshare.com/wiki/RS485_TO_ETH_(B)
Raspberry Pi 3 and Home Assistant running in container
Nilan integration installed from HACS

When trying to add integration, I get 'Failed to connect' error:
[image]

I've managed to set up my waveshare converter (or at least I thinks so):
[image]
[image]

Loggin gives following:
'2022-12-20 00:14:37.435 ERROR (MainThread) [pymodbus.client.sync] Connection to (192.168.1.200, 502) failed: timed out'

I'm quite sure that this has something to do with my waveshare configs? First I thought that this problem is caused by running HA in container and mapping not workin between container and host. But then found out that Host <-> HA running in container have all the ports opened as a 'default' I guess. 

When loggin in to my router settings, I can see waveshare IP in the ARP/RARP Table.

Perhaps some of you who has managed to get this integration work with waveshare could help. All the help is highly appreciated and hopefully even rewarded in the end.

veista, 2022-12-19:
Hi, failed to connect means that no data was exchanged. Check if you can ping it through HA.

Magifi, 2022-12-19:
Did that on host (raspberry os) and no answer. As well my laptop couldn’t get answer. So the problem must be with the converter network settings. Too tired to go on, have to continue tomorrow if time schedules allow. Thanks for the help 👍

MKBU2, 2022-12-20:
Configuration is identical to mine and for me it works. Sharewave and Raspberry HA are connected to same switch.

Magifi, 2022-12-20:
Well, some time it's good to take a break and continue later. During the day I was trying to figure out what could be the problem. Double checking the ip addresses and stupid mistake was found. Third chapter of ip should have been 0 but I typed 1. So changing those three ip adresses in VirCom configuration tool made it work and I could connect Nilan with no problems.

There's some strange behaviour still goin on. Temperatures and other items are jumping like crazy and some times shows 'unknown'. I also had to remove and reinstall integration before anything showed up. I'm not sure but perhaps my other raspberry running openHAB parallel to this could cause some weird stuff. It's still stable and shows everything like before. But this is one step forward and great approvement! Thanks for all the help.

[image]
[image]
[image]

Magifi, 2022-12-20:
Well that was a short run. Don't know what I messed (perhaps configurations.yaml and logger rows were bad). So I deleted those rows and restarted HA (couple of times). Now nothing and this is what I found from Nilan logs in the frontend:

2022-12-20 23:33:20.249 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Nilan for nilan
File "/config/custom_components/nilan/__init__.py", line 39, in async_setup_entry
File "/config/custom_components/nilan/device.py", line 60, in setup
File "/config/custom_components/nilan/device.py", line 124, in get_machine_type

My own fault but you learn from your mistakes, right? I know this is not topic related but perhaps someone else can get few tips of these posts.. Like I have done thousand times when faced a problem and have started googling around.

veista, 2022-12-20:
What is the question? I think you should remove the other device from the modbus, you can NOT have multiple hosts on modbus RTU since this creates a competition conflict. This error comes from not being able to read the register.

veista, 2022-12-20:
If your Modbus bridge supports multiple TCP masters you could add your other solution like that, provided they have different IP addresses.

Magifi, 2022-12-20:
> What is the question? I think you should remove the other device from the modbus, you can NOT have multiple hosts on modbus RTU since this creates a competition conflict. This error comes from not being able to read the register.

My bad (again). I thought that Nilan controller is the master and other devices are slaves (like local control panel + other devices connected via modbus). So my assumption was that there could be more than one device and all could run parallel at the same time causing no problem to each other. I will remove my other Raspberry with RS485 - usb converter as soon as I’m familiar enough with HA.

Mistakes happen and learning takes time..

veista, 2022-12-20:
And By-the-way you should check the validity of your configuration.yaml from developer tools 
[image]

closed 2022-12-21 by veista