veista/nilan #24: Please add T4 start de-icing support for Combi 302 Polar

abstacek, 2022-12-20:
Please add the Combi 302 Polar function in the HMI De-icing T4 start de-icing option to set ON/OFF. Thank you.
[image]

veista, 2022-12-20:
I believe it is already there "Pre-Heating Temperature"

abstacek, 2022-12-20:
You may have deleted it. I only have Pre-Heating Effect.

veista, 2022-12-20:
Ok, I will check.

veista, 2022-12-20:
Hi. Yes there is a typo, will fix in next release.

veista, 2022-12-20:
Hi, Even though there was a typo, it should not affect this. You should have this entity. "Pre-Heating Temperature"

abstacek, 2022-12-20:
I don't really have this entity. I have 76 entities but this one is missing.
[image]

veista, 2022-12-20:
Yes, there we two mistakes, deleted the previous comment.

closed 2022-12-20 by veista