veista/nilan #25: 'Invalid response from device'

AlvinKleine, 2022-12-20:
Trying to create serial connection to Nilan, but getting error "invalid response from device".

[image]

Setup
- HA on Raspberry PI 4 
- CTI USB-Nano-485 Interface Converter (connected to HA USB0)
- Nilan CTS602 with SW 2.38

...any thoughts?

Br
Beginner

veista, 2022-12-20:
Most likely cause is wrong device path.

veista, 2022-12-20:
Go to setting->system->hardware->"three dots"->all hardware

Find the correct usb adapter. Copy path.

veista, 2022-12-21:
Missing ground wire could also be one cause.

AlvinKleine, 2022-12-21:
Thank You! Issue solved. I had physical wire connection problem.

veista, 2022-12-21:
> Thank You! Issue solved. I had physical wire connection problem.

You are most welcome.

closed 2022-12-21 by veista

veista, 2022-12-21:
I would be interested in your device bus version. If you could kindly report it here #9. Thanks.

AlvinKleine, 2022-12-21:
> I would be interested in your device bus version. If you could kindly report it here #9. Thanks.
reported. it's 19.

Should one still be able to run following type of holding register services even your perfect HACS is active? Looking to build Nordpool sensor triggered automations w/o python skills, but e.g. below service does not turn PILP off:

service: modbus.write_register
data:
  address: 1001
  slave: 30
  value: 0
  hub: hub1