veista/nilan #35: Unsupported Device: VPL 15

janner, 2023-01-10:
Debug log:
> 2023-01-10 18:35:09.225 DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 4 not found in supported devices list

Attached is a picture of the plate.
[image]

veista, 2023-01-10:
Thanks. Could you also take a picture of the menu item requested.

veista, 2023-01-10:
Added support.

janner, 2023-01-11:
This is the non-HMI CTS602, so I'm guessing this is the information you are looking for.
[image]
[image]
[image]

veista, 2023-01-11:
Yes, thank you very much. Will fix the type on next release, so you will have to remove and add the integration again.

closed 2023-01-12 by veista