veista/nilan #38: Support for VPL 25

netforceatg, 2023-01-11:
Hello,

I have a VPL25c with a CTS602 board with old push button display. I believe this is very similar to VPL 15c which is already supported by this integration.

I replaced the CTS600 with a CTS602 - 512 motherboard back in 2021, to get Modbus interface. The board runs sw version 2.37.C. I've used a Modbus TCP adapter by the name of Hi-flying elfin EE11 (RS485 to ModBus TCP) to control VPL25c manually up until this point and has been working fine. But I would very much get it nicely integrated to Home Assistant

When I try to add it this integration to home assistant with the ip address it says device not supported. How can I help to get this VPL28c supported by this integration?

veista, 2023-01-11:
Hi,

Follow the instructions from read me.

netforceatg, 2023-01-11:
Hi,

Sorry for not posting this the last time.

**Log:**
2023-01-11 14:53:58.351 DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 25 not found in supported devices list

**QModMaster reveals the following:**
Bus.Version: 11
App.VersionMajor: 2.
App.VersionMinor: 37
App.VersionRelease: c

**On CTS display :**
TYPE: VPL 25c

[image]

veista, 2023-01-11:
Added support

netforceatg, 2023-01-11:
Wow, that was fast. And I just tested it, seems to be working great! 

Thank you 🙏 

I’ll post the picture of the panel next time I’m at the location.

closed 2023-01-11 by veista