veista/nilan #42: Error importing config flow

gllmlbrt, 2023-01-26:
Hello, 

Wanted to take the fresh 2022.2.0b1 today, after installing it i get the following error:
```
Logger: homeassistant.config_entries
Source: config_entries.py:1211
First occurred: 08:33:14 (1 occurrences)
Last logged: 08:33:14

Error importing platform config_flow from integration nilan to set up nilan configuration entry: No module named 'pymodbus.client.sync'
```
So the integration does not load.

Tried redownloading the custom integration, and restart again, but not improvement.

Do you know of a major change of pymodbus in the upcoming HA updates ?
I  could find similar issues reported in earlier upgrade of HA here:
[https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/162](url)

And I can see that in 2022.2, core pymodbus bumped to 3.1.1.
[https://github.com/home-assistant/core/pull/86513](url)
I will try to change that in the manifest and see if that works.

Not sure if it is the same reason. Also home assistant now requires python 3.10 minimum, not sure if related.

netforceatg, 2023-02-02:
After I upgraded to 2023.2.0 I got the same issue

Also trie to remove the plugion from HACS, and reinstalling it. When loading it as an integration in home assistant I get the following error message:

Error
Config flow could not be loaded: {"message":"Invalid handler specified"}

veista, 2023-02-02:
I will look into this issue this week. It's probably an issue with the manifest file.

salosa, 2023-02-02:
Simple fix is to replace module 'pymodbus.client.sync' with 'pymodbus.client.' in config_flow.py

gllmlbrt, 2023-02-03:
> Simple fix is to replace module 'pymodbus.client.sync' with 'pymodbus.client' in config_flow.py [image]

I can confirm this fixes the issue.

closed 2023-02-03 by veista