veista/nilan #52: Unsupported device (Comfort 300 LR Light HMI/Waveshare RS485 to ETH (B)

JakobPP, 2023-03-07:
Hello

Thanks for your great work, I am very much looking forward to finally getting my HVAC online at some point.

I've been struggling with getting the plugin to connect with my Comfort 300 however. I've already been over all the wiring, termination etc. I have an powermeter online on the bus as well (PME3250) and I've measured correct voltage levels and "activity" on connection terminals at the plug on the HVAC unit itself. So, im fairly confident that this issue has to solved on "the software side".
Im using the waveshare, and pasting in my config at the end just for reference.

I hope the info supplied below is what you need, or else please let know. Thanks again.

HA debug log when trying to configure the plugin:

2023-03-07 21:07:09.339 DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 2 not found in supported devices list

Modelplate:

[image]

HMI:

[image]

Waveshare config (for reference):

[image]

veista, 2023-03-07:
Great! I will add this ASAP, by the end of the week.

JakobPP, 2023-03-08:
Excellent, Thanks!Venlig hilsen - - Best regardsJakob Pilgaard Pedersen

veista, 2023-03-09:
The device is now added.

closed 2023-03-09 by veista

JakobPP, 2023-03-21:
Thanks once again veista for quick response. Its been running fine since then. I just wanted to check back in, in case anyone else is looking for help with a similar solution using the Waveshare.

Its possible to use the gatway in Modbus TCP server mode as well.

Posting screenshot for reference.

[image]