veista/nilan #55: Integrating VP18cCom via RS485 serial USB

whoami-pi, 2023-03-11:
Hi,

I'm trying to connect my Nilan VP18cCom via RS485 serial USB with HA (using also CTS602).
I'm getting the error message "Unsupported Device" during the integration of Nilan.

I'm new at HA, the same interface was working before via PI and Node Red.
I've added this into configration.yaml and restarted HA:
`# Logger
logger:
  default: critical
  logs:
    custom_components.nilan: debug
    custom_components.hacs: debug`

But I'm still unable to see any log entries in Log-book. Have also activated debug protocoll for the HACS device.
Here is my selection:
[image]

As the path seems to be correct:
[image]

veista, 2023-03-11:
You should get an entry in logbook after trying to configure the device. Also provide the pictures asked in read-me.

whoami-pi, 2023-03-12:
Hi

thanks for your quick response. Please find attached pics from Nilan devices.

Thanks for checking!!

[image]
[image]
[image]
[image]

whoami-pi, 2023-03-12:
This is actually where I stuck up. So, what I did:
1. Debug is active for HACS:
[image]
[image]
2. Configuration.yaml: Have added these lines, saved the file, restarted HA
`# Logger`
`logger:`
`  default: info`
`  logs:`
`    custom_components.nilan: debug`
`    custom_components.hacs: debug`
3. (After new start STRG & F5 for cache) Add new integration, entered Nilan, selected SERIAL and following
[image]

4. Getting the issue related to unsupported device
5. Checking the Logbook, no entry for this integration
6. Even in home-assistant.log no entry related to this:
[image]

I'm pretty sure that I'm doing something wrong...

veista, 2023-03-12:
That should work. You don't need the HACS debugging.

whoami-pi, 2023-03-12:
But still not able to see any entries in Log Book for Nilan or am I checking the wrong log book?
[image]

veista, 2023-03-12:
> But still not able to see any entries in Log Book for Nilan or am I checking the wrong log book? [image]

Yes. Go to settings->system->logs.

whoami-pi, 2023-03-12:
Thanks, as I said, newbie at HA, here is an entry for Nilan, does it help you?

Logger: homeassistant.util.async_
Source: util/async_.py:159
First occurred: 10:35:30 (3 occurrences)
Last logged: 10:35:30

Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for nilan doing blocking calls at custom_components/nilan/config_flow.py, line 59: result = client.read_holding_registers(

veista, 2023-03-12:
DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type XX not found in supported devices list

I meed this entry from logs.

whoami-pi, 2023-03-12:
ok, now it should be the right one, hopefully, thanks for being patient with me :)

DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 12 not found in supported devices list

veista, 2023-03-12:
Great. I will add your device ASAP.

veista, 2023-03-12:
Support for your device is now in release 1.1.17

closed 2023-03-12 by veista