veista/nilan #62: Concurrent connections on ModbusRTU from different integrations - read errors

jires, 2023-07-25:
I'm not sure if this is a issue with the integration or with HA DataUpdateCoordinator, so clarifying will be helpful.

I have a one Modbus serial bus, with multiple devices on it. It's connected to HA via USB-RS485 adapter.
There is 2 integrations which using Modbus RTU, one is Nilan, the other is native Modbus Integration built in a HA which reads all other devices on the bus.
Update interval for both is set to 30 sek.

When HA starts and run, both integrations tries to read registers from different devices at the same time. It resulting in communications errors and unavailable sensors.  

My workaround for this is to disable Nilan integration manually and observe the led's on usb adapter. When led's stops flashing (native Modbus integration is done with serial bus for now), I'm enabling Nilan after few seconds.  In that way I have those 2 integrations shifted in time with theirs readings, and everything working well.... until next HA restart.

veista, 2023-07-25:
Yes. It is a complicated problem. The main issue is the lack of capabilities of your modbus tcp bridge and/or too slow end devices. There is no clever solution for this. The problem is that you now have two different connections to the bridge and it does not handle it properly.

Trying a different bridge device might solve the problem. Or creating your own. Best solution is to have separate bridge devices so that there is only one connection per modbus network.

Does the device hang completely? Does not anwser to either connection?

veista, 2023-07-26:
Sorry, misread at first. It is surprising it lets you connect to the latter when using a USB adapter. The problem is even more severe when using USB since the COM port only supports one instance. Easier is to use a separate USB adapter.

jires, 2023-07-26:
Yes, I use USB adapter, [this one.](https://www.amazon.de/-/en/Industrial-USB-RS485-Converter-300-921600bps/dp/B082D5PXJ4/)
Sometimes, when HA starts, Nilan integration fail to start and I need to reload it. Sometimes is starting well but lot of entities for both integrations (the native one and Nilan) goes "Unknown" on and off,  and spits errors in log:

``
2023-07-25 20:31:20.253 ERROR (MainThread) [custom_components.nilan.device] Could not read get_ventilation_state
2023-07-25 20:31:20.349 ERROR (SyncWorker_14) [homeassistant.components.modbus.modbus] Pymodbus: nilanhub: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
2023-07-25 20:31:20.429 ERROR (MainThread) [custom_components.nilan.device] Could not read get_supply_fan_level
2023-07-25 20:31:20.430 ERROR (MainThread) [homeassistant.helpers.entity] Update for climate.nilan_hvac fails
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 695, in async_update_ha_state
await self.async_device_update()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 940, in async_device_update
await self.async_update()
File "/config/custom_components/nilan/climate.py", line 183, in async_update
elif fan_supply_level > 0:
^^^^^^^^^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'NoneType' and 'int'
2023-07-25 20:31:45.470 WARNING (SyncWorker_19) [pymodbus.logging] Cleanup recv buffer before send: 0x1e 0x3 0x2 0x0 0x2 0xac 0x47
2023-07-25 20:31:45.502 ERROR (SyncWorker_11) [homeassistant.components.modbus.modbus] Pymodbus: Nilan: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
2023-07-25 20:31:45.534 ERROR (MainThread) [custom_components.nilan.device] Could not read get_ventilation_step
2023-07-25 20:31:45.622 ERROR (SyncWorker_19) [homeassistant.components.modbus.modbus] Pymodbus: nilanhub: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
2023-07-25 20:31:45.703 ERROR (MainThread) [custom_components.nilan.device] Could not read get_control_state
2023-07-25 20:31:48.392 ERROR (SyncWorker_8) [homeassistant.components.modbus.modbus] Pymodbus: Nilan: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
2023-07-25 20:31:48.424 ERROR (MainThread) [custom_components.nilan.device] Could not read get_supply_fan_level
``
Besides Nilan, I have 28 more entities on native integration. When the issue is happening, HA can get maybe 10 values read per update, so is not completely fails.

Using different adapters is not a option for me, house is wired with one bus only, and devices are far away.

Before I installed Nilan integration, everything was connected with native Modbus integration, (with name nilanhub, therefore it show up in logs), but your integration is far more advanced than this I had wrote down in yaml by myself, so I started using yours.

Maybe I should write an enhancement "wish" on HA core Github, so those smart guys can implement some kind of coordinator for pymodbus, when there are coming calls from different integrations on the same physical device, it can be  queued somehow..