veista/nilan #67: Translation Help

veista, 2023-07-29:
If you are a native speaker of any language not yet supported or want to improve the translations, please do so and leave a pull request.

You can use the EN.json Translation File as a template from the dev branch.

Thank You!

Takethis88, 2023-09-22:
I'm working on the Italian translation. Just waiting to test it on my side before committing it. I'm saying this to avoid duplicating work.

arjanmanders, 2024-06-28:
[Dutch translation!](https://github.com/user-attachments/files/16027705/nl.json)

Pepekliste, 2025-02-07:
Czech (quick) translation - comments welcome

[cs.json](https://github.com/user-attachments/files/18709727/cs.json)

Pepekliste, 2025-09-17:
> Czech (quick) translation - comments welcome
> 
> [cs.json](https://github.com/user-attachments/files/18709727/cs.json)

I am sending the revised and checked version. Would it be possible to add it to the default available translations in the add-on? It is rather annoying to have to reapply it manually after each update.

[cs.json](https://github.com/user-attachments/files/22377197/cs.json)