veista/nilan #70: Upgrade to 2023.7 core breaks Nilan connection

ollila, 2023-07-31:
I have Nilan EC9 with CT602 mobus v22 connected to HA (Core 2023.6.3) through Waveshare RS485-to-Eth working as intended. When I upgrade to 2023.7 series core something breaks and connection between HA and Nilan is lost. The issue is replicable and connection returns when reverting to 2023.6 core from backups.

veista, 2023-07-31:
Hi, Which version of the integration are you running?

ollila, 2023-07-31:
I'm currently running 1.1.17

veista, 2023-07-31:
Have you tried the beta? I'm running that with newest HA and it is running fine on my setup.

ollila, 2023-08-01:
I tried 1.2.1 and managed to get it running under 2023.6 (although I messed sensor names). When updating to 2023.7.3 it loses connection to Nilan. From HA logs I see the following error messages:

Could not read get_bus_version
6:21:33 PM – (ERROR) Nilan (custom integration)
Could not read get_machine_type
6:21:31 PM – (ERROR) Nilan (custom integration)
Pymodbus: Nilan: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
6:21:31 PM – (ERROR) Modbus

Since pymodbus was updated in 2023.7 and produces error here I would guess these are linked.

veista, 2023-08-01:
Have you tried restarting the bridge device after updating?

ollila, 2023-08-01:
I will try it next but it will most likely slip to weekend before I have enough time in case I need to restore installation from back up.

ollila, 2023-08-04:
I tried the bridge restart without success this time with 2023.8 core. After restart bridge has only the link light on (green) but not the blue indicating bridge has not connected to host. I enabled debug logging before update and the only usefult messages are what I already saw from HA logs:

2023-08-04 20:25:48.153 ERROR (SyncWorker_4) [homeassistant.components.modbus.modbus] Pymodbus: Nilan: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
2023-08-04 20:25:49.511 ERROR (MainThread) [custom_components.nilan.device] Could not read get_machine_type
2023-08-04 20:25:49.603 ERROR (MainThread) [custom_components.nilan.device] Could not read get_bus_version

Here is the bridge configuration in case it makes any difference:
[image]

veista, 2023-08-04:
I don't have this bridge my self but your HA instance and the bridge are on different subnets. And work mode should be server, not client.