veista/nilan #74: Unknown Device - Combi Polar 302 HMI

jamespreedy, 2023-08-13:
Hey there, thanks for your work on this integration.

I've tried to install, and I'm getting a device not supported error.

Device type plate below.

[image]

Photo showing "Show Data => Type" from CTS602 controller below.

[image]

Believe from a previous post that this is what you need from the debug logs? If you need more, please let me know - will be very happy to help.

`2023-08-13 21:13:26.622 DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 33 not found in supported devices list`

Thanks in advance!

veista, 2023-08-13:
Great, thanks! I will add this ASAP.

jamespreedy, 2023-08-13:
@veista - thanks for the speedy response. Really appreciate it! Have a great week. 🍺

closed 2023-08-14 by veista