veista/nilan #76: nilan: Error on device update!

jamespreedy, 2023-08-18:
@veista - thanks so much for adding support for my Nilan. Got it all up and running, nice dashboard etc - great stuff!

I get 8x errors in the logs on startup. I seem to be getting other data through ok, but wanted to let you know.

Seems to be related to the below ...
get_supply_heating_pid_time
get_compressor_stop_time
get_low_temperature_compressor_start_setpoint
get_max_return_step
get_min_supply_step
get_central_heat_type
get_central_heat_select
get_air_heat_select

I don't seem to have similarly named entities created.

Please let me know if I can help, or get further logs etc etc - very happy to help! 😀

```
Logger: homeassistant.components.number
Source: custom_components/nilan/device.py:1438 
Integration: Number (documentation, issues) 
First occurred: 6:23:02 am (1 occurrences) 
Last logged: 6:23:02 am

nilan: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 576, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 938, in async_device_update
    await self.async_update()
  File "/config/custom_components/nilan/number.py", line 579, in async_update
    self._attr_native_value = await getattr(self._device, self._attribute)()
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/nilan/device.py", line 1438, in get_supply_heating_pid_time
    result.registers[0].to_bytes(2, "little", signed=False),
    ~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range

Logger: homeassistant.components.number
Source: custom_components/nilan/device.py:1528 
Integration: Number (documentation, issues) 
First occurred: 6:23:02 am (1 occurrences) 
Last logged: 6:23:02 am

nilan: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 576, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 938, in async_device_update
    await self.async_update()
  File "/config/custom_components/nilan/number.py", line 579, in async_update
    self._attr_native_value = await getattr(self._device, self._attribute)()
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/nilan/device.py", line 1528, in get_compressor_stop_time
    result.registers[0].to_bytes(2, "little", signed=False),
    ~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range

Logger: homeassistant.components.number
Source: custom_components/nilan/device.py:1124 
Integration: Number (documentation, issues) 
First occurred: 6:23:02 am (1 occurrences) 
Last logged: 6:23:02 am

nilan: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 576, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 938, in async_device_update
[... 127 lines cut ...]
IndexError: list index out of range

Logger: homeassistant.components.select
Source: custom_components/nilan/device.py:167 
Integration: Select (documentation, issues) 
First occurred: 6:23:02 am (1 occurrences) 
Last logged: 6:23:02 am

nilan: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 576, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 938, in async_device_update
    await self.async_update()
  File "/config/custom_components/nilan/select.py", line 286, in async_update
    self._attr_current_option = str(await getattr(self._device, self._attribute)())
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/nilan/device.py", line 167, in get_air_heat_select
    result.registers[0].to_bytes(2, "little", signed=False),
    ~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range

```

`

veista, 2023-08-18:
Great, could you give me your modbus version?

jamespreedy, 2023-08-18:
Sorry, forgot that bit - version 21. Thanks!

veista, 2023-08-23:
Seems like the issue is with modbus version 21 devices. The registers are there, but don't return any value. It is not the first time modbus version 21 is being a problem.

I'm on a business trip so resolving this will take time.

jamespreedy, 2023-08-23:
No unavailable entities, and no entities named something like the list I added above - it maybe *seems* that it tries to create them and couldn't, so errored?

Thanks for your help with this one - appreciate it!

jypma, 2023-08-23:
I've encoutered the same issue (see #78). I've traced the uncreated entities to these register definitions:
```
CTS602HoldingRegisters.air_temp_heat_select      1209
CTS602HoldingRegisters.central_heat_heat_select  1801
CTS602HoldingRegisters.central_heat_heat_type    1807
CTS602HoldingRegisters.air_flow_inlet_min        4015
CTS602HoldingRegisters.air_flow_exhaust_min      4016
CTS602HoldingRegisters.air_flow_exhaust_max      4017
CTS602HoldingRegisters.air_temp_temp_min_cpr     1502
CTS602HoldingRegisters.air_temp_cpr_restart      1503
CTS602HoldingRegisters.central_heat_reg_time     1808
```
They all seem to be (newer?) registers that aren't present on our devices (or firmware versions). A quick fix could be of course to just in `device.py` change all of the `if result is not None` to sth like `if result is not None and len(result.registers) > 0`.

Alternatively, that check could pushed down into a small helper function that wraps `self._modbus.async_pymodbus_call`, so it need not be repeated all over.

veista, 2023-08-23:
The problem is either: 1. Your device model does not support these functions, which is an easy fix or 2. Your modbus version does not support these functions, which is more complicated since...

... I designed the software thinking register capabilities are compounding compared to modbus version. Modbus 20 and 21 seem to be outliers since it is where Nilan changed to HMI panel type. 19 is the newest for the old panels and 22 is the newest ive encountered for the touch panel. 19 and 22 have pretty much all functionality available.

This will take time to figure out since Im on a long business trip and dont have access to a developement environment.

veista, 2023-08-23:
Also modbus should not be able to return an empty register. This might be an issue with the newer pymodbus which also needs investigation.

It should give a register not found error.

jypma, 2023-08-23:
Fair enough :)

I might be leaning towards your explanation (2), since a cursory glance at the option that are available in the (user) menu on my touch control panel don't really relate to the register names here. However, they might be locked under factory/installer settings though, which has some sort of "password" on it that I don't know... and the guys that installed it years ago of course no longer exist... :see_no_evil: 

Either way, feel free to let it sit for a while, since the important part is working. The integration accepts changes fine, too (now automating Nilan to do extra cooling when our solar cells have excess power).

veista, 2023-08-23:
On cts602 no registers require a password. Only if you access them via the hmi panel. The password for service settings is 2. 

I cannot remember the installer password but it was written in some forum. There is nothing you should change so better to not go there, device breaking stuff.

closed 2023-09-22 by veista