veista/nilan #77: After Upgrade Re-Installation raises an error

whoami-pi, 2023-08-21:
Hi,

have upgraded to version 1.2.3 running on HA 10.5 with 2023.8.3. 

I've Nilan integrated via Serial, which was working now for a while.

Now after the HACS Upgrade, restart of HA and unplug/plug of Nilan USB, I've clicked on "Add Integration" and have selected the   Interface type "Serial".

Normally next step is the selection of the usb path but now the system is throwing a popup with title "Fehler" without any further information.
[image]

Enabling log, restart HA and trying same to find this in the system log.

Any idea how to solve it?

`Logger: aiohttp.server
Source: data_entry_flow.py:398 
First occurred: 19:24:21 (1 occurrences) 
Last logged: 19:24:21

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 148, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 181, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 292, in async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 398, in _async_handle_step
    if not isinstance(result["type"], FlowResultType):
                      ~~~~~~^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
`

veista, 2023-08-22:
Hi, that is odd. Can you test if version 1.1.x works?

whoami-pi, 2023-08-22:
> Hi, that is odd. Can you test if version 1.1.x works?

Hi, thanks for your quick response. Going back to version 1.1.19 works perfectly again.

jires, 2023-08-22:
I encountered same issue. Revert to 1.1.19 last working version.

veista, 2023-08-23:
This issue should now be resolved in 1.2.4. I don't know how but an indentation got slipped into the serial part breaking it. It was a simple fix.