veista/nilan #78: Fresh install of 1.2.3 yields "IndexError: list index out of range" when accessing registers

jypma, 2023-08-22:
I've done a fresh install of version 1.2.3 into my Home Assistant 2023.7.3. Installed by unzipping the Github release zip file. Communicating with a COMBI 302 T over Modbus-TCP.

The installation itself went fine from a UI perspective (no errors reported, although all the detailed sensors took a while to appear). However, there are a few `IndexError` entries in the log:

```
2023-08-22 07:37:05.541 INFO (MainThread) [homeassistant.setup] Setting up nilan
2023-08-22 07:37:05.541 INFO (MainThread) [homeassistant.setup] Setup of domain nilan took 0.0 seconds
2023-08-22 07:37:05.614 INFO (SyncWorker_20) [homeassistant.components.modbus.modbus] modbus Nilan communication open
2023-08-22 07:37:05.950 INFO (MainThread) [homeassistant.components.button] Setting up button.nilan
2023-08-22 07:37:05.952 INFO (MainThread) [homeassistant.components.climate] Setting up climate.nilan
2023-08-22 07:37:05.954 INFO (MainThread) [homeassistant.components.water_heater] Setting up water_heater.nilan
2023-08-22 07:37:05.957 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.nilan
2023-08-22 07:37:05.959 INFO (MainThread) [homeassistant.components.select] Setting up select.nilan
2023-08-22 07:37:05.961 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.nilan
2023-08-22 07:37:05.964 INFO (MainThread) [homeassistant.components.number] Setting up number.nilan
2023-08-22 07:37:05.965 INFO (MainThread) [homeassistant.components.switch] Setting up switch.nilan
2023-08-22 07:37:05.969 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new button.nilan entity: button.ni
lan_synchronize_time_and_date
2023-08-22 07:37:06.107 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_modbus_version
2023-08-22 07:37:06.177 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_controller_board_temperature_t0
2023-08-22 07:37:06.246 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_fresh_air_intake_temperature_t1
2023-08-22 07:37:06.313 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_waste_air_temperature_t4
2023-08-22 07:37:06.379 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_condenser_temperature_t5
2023-08-22 07:37:06.447 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_waste_air_temperature_t6
2023-08-22 07:37:06.516 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_supply_air_temperature_t7
2023-08-22 07:37:06.591 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_return_air_temperature_t10
2023-08-22 07:37:06.664 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_user_panel_temperature_t15
2023-08-22 07:37:06.731 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_humidity
2023-08-22 07:37:06.796 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_alarms_active
2023-08-22 07:37:06.865 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_control_state
2023-08-22 07:37:06.931 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_time_in_control_state
2023-08-22 07:37:06.999 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_supply_fan_level
2023-08-22 07:37:07.077 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_return_fan_level
2023-08-22 07:37:07.143 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_days_since_air_filter_change
2023-08-22 07:37:07.209 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.ni
lan_days_to_air_filter_change
2023-08-22 07:37:07.275 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.nilan_climate_season
2023-08-22 07:37:07.341 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.nilan_exchanger_efficiency
2023-08-22 07:37:07.410 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.nilan_24h_average_humidity
2023-08-22 07:37:07.477 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.nilan_ventilation_state
2023-08-22 07:37:07.544 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.nilan_return_fan_speed
2023-08-22 07:37:07.610 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.nilan_supply_fan_speed
2023-08-22 07:37:07.684 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.nilan_time
2023-08-22 07:37:07.751 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.nilan entity: sensor.nilan_after_heating_type
2023-08-22 07:37:07.818 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new select.nilan entity: select.nilan_cooling_mode_ventilation_level
2023-08-22 07:37:07.898 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new select.nilan entity: select.nilan_air_filter_alarm_interval
2023-08-22 07:37:07.964 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new select.nilan entity: select.nilan_cooling_setpoint_offset
2023-08-22 07:37:08.030 ERROR (MainThread) [homeassistant.components.select] nilan: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 570, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 940, in async_device_update
    await self.async_update()
  File "/config/custom_components/nilan/select.py", line 286, in async_update
[... 123 lines cut ...]
    self._attr_native_value = await getattr(self._device, self._attribute)()
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/nilan/device.py", line 1438, in get_supply_heating_pid_time
    result.registers[0].to_bytes(2, "little", signed=False),
    ~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2023-08-22 07:37:09.801 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_maximum_time_in_high_humidity_ventilation
2023-08-22 07:37:09.869 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_low_outdoor_temp_setpoint
2023-08-22 07:37:09.937 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_supply_heater_delay
2023-08-22 07:37:10.004 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_low_room_temperature_setpoint
2023-08-22 07:37:10.073 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_defrost_start_setpoint
2023-08-22 07:37:10.140 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_defrost_stop_setpoint
2023-08-22 07:37:10.215 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_supply_fan_power_at_level_1
2023-08-22 07:37:10.283 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_supply_fan_power_at_level_2
2023-08-22 07:37:10.351 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_supply_fan_power_at_level_3
2023-08-22 07:37:10.424 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_supply_fan_power_at_level_4
[... cut ...]
2023-08-22 07:37:10.704 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.nilan entity: number.nilan_return_fan_power_at_level_4
2023-08-22 07:37:10.974 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new select.nilan entity: select.nilan_reset_alarm
2023-08-22 07:37:11.458 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new climate.nilan entity: climate.nilan_hvac
```

Please let me know if there's anything I can do to help investigate further. Thanks for a great integration so far!

veista, 2023-08-22:
Please submit Modbus Version, Device Type - as Shown in the Integration, Device Version - as Shown in the Integration

jypma, 2023-08-22:
"Modbus Version" says 21. I don't have anything exactly called "Device Type" nor "Device Version".  However, under "Device Info" it says:
```
COMBI 302 T
by Nilan
Firmware: 1.6.26
Hardware: 16 
```

veista, 2023-08-23:
This is related to issue #76

closed 2023-08-23 by jypma