veista/nilan #79: All sensors show same name

joriws, 2023-08-26:
Not nice - is this a feature of this integration?

[image]

entity ids for adding sensor to cards are like:
[image]

joriws, 2023-08-26:
I deleted integration and re-created, the same thing. HACS does not say update is available. I am running older core

[image]

[image]

veista, 2023-08-26:
My bad partially. I forgot to change minimum HA version .

veista, 2023-09-16:
Will close this now. Sorry I can't support every HA version/device combination. You can try older releases and hope it has your device supported. I will be more mindfull of updating the HA requirement before release. This will not prevent misuse if installed manually though.

closed 2023-09-16 by veista