veista/nilan #80: 'ModbusHub' object has no attribute 'async_pymodbus_call'

gllmlbrt, 2023-08-31:
Hello, just updated to 2023.09.b0 to test and got the below error. I am unable to clearly determine what the core issue is, but perhaps you can ?

Logger: homeassistant.config_entries
Source: custom_components/nilan/device.py:120
Integration: Nilan (documentation, issues)
First occurred: 10:50:22 (1 occurrences)
Last logged: 10:50:22

Error setting up entry Nilan for nilan
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/nilan/__init__.py", line 41, in async_setup_entry
    await device.setup()
  File "/config/custom_components/nilan/device.py", line 61, in setup
    hw_type = await self.get_machine_type()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/nilan/device.py", line 120, in get_machine_type
    result = await self._modbus.async_pymodbus_call(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ModbusHub' object has no attribute 'async_pymodbus_call'

veista, 2023-09-07:
Hi,

Can't replicate this with dev0, on 2023.09 yes. I suspect this issue is already fixed on HA side.

veista, 2023-09-07:
https://github.com/home-assistant/core/issues/99784

veista, 2023-09-07:
https://github.com/home-assistant/core/pull/97780/files

veista, 2023-09-07:
I can not fix this issue fully right now. We will have to wait for an update for HA. The pymodbus wrapper in HA is broken. If this is not fixed I will have to make my own, but not looking forward to that.

Will push my progress to dev, so you can try it yourself.

jamespreedy, 2023-09-08:
@veista - thank you! 🍺

kfroeschl, 2023-09-08:
hi,

Is the problem not related to a name-change in the ha-modbus wrapper-lib?
https://github.com/home-assistant/core/commit/bbc34bae87d2c5fac4e1bb89831dd427ccc8c2c1#diff-2665704d3ee341f20f11d81ce07da5575297bf5c2dc1cb6623c92d6f62ed8e1a

veista, 2023-09-08:
> hi,
> 
> Is the problem not related to a name-change in the ha-modbus wrapper-lib?
> https://github.com/home-assistant/core/commit/bbc34bae87d2c5fac4e1bb89831dd427ccc8c2c1#diff-2665704d3ee341f20f11d81ce07da5575297bf5c2dc1cb6623c92d6f62ed8e1a

Concider reading the issue. Yes that is one problem. You can download the fix for that from dev. Even though I implemented that change, it does not work. I spent a couple of hours debugging and figured that the new pymodbus version is not fully compatible with the wrapper used in HA.

veista, 2023-09-08:
> @veista - thank you! 🍺

Than you for your continuing support!

veista, 2023-09-08:
> hi,
> 
> Is the problem not related to a name-change in the ha-modbus wrapper-lib? [home-assistant/core@bbc34ba#diff-2665704d3ee341f20f11d81ce07da5575297bf5c2dc1cb6623c92d6f62ed8e1a](https://github.com/home-assistant/core/commit/bbc34bae87d2c5fac4e1bb89831dd427ccc8c2c1#diff-2665704d3ee341f20f11d81ce07da5575297bf5c2dc1cb6623c92d6f62ed8e1a)
> 
> I've replaced it locally in device.py async_pymodbus_call -> async_pb_call and now all entities are there again and updating
> 
> No exceptions in the HA Logs either so far

Are you running with Modbus tcp or serial?

kfroeschl, 2023-09-08:
Modbus Serial

I'll monitor it for some time if it stops working and maybe add the suggested error-handling arguments
( retries: 3, retry_on_empty: true, close_comm_on_error: false )

@veista - great job btw. I love this integration 🍺

veista, 2023-09-08:
> Modbus Serial
> 
> I'll monitor it for some time if it stops working and maybe add the suggested error-handling arguments ( retries: 3, retry_on_empty: true, close_comm_on_error: false )
> 
> @veista - great job btw. I love this integration 🍺

OK. Then it seems to me that only the TCP side is broken or it depends also on the bridge device.

veista, 2023-09-08:
Can you test the dev branch if it works for you?

kfroeschl, 2023-09-08:
"dev" branch did not work unfortunately
```
DEBUG (MainThread) [pymodbus.logging] Connecting to /dev/ttyUSB0.
DEBUG (MainThread) [pymodbus.logging] Connecting comm
DEBUG (MainThread) [pymodbus.logging] Connected to comm
DEBUG (MainThread) [pymodbus.logging] callback_connected called
DEBUG (MainThread) [pymodbus.logging] send: 0x1e 0x3 0x3 0xe8 0x0 0x1 0x6 0x15
DEBUG (MainThread) [pymodbus.logging] Adding transaction 30
DEBUG (MainThread) [pymodbus.logging] recv: 0x1e 0x3 0x2 0x0 0x23 0x6c 0x5f addr=None
DEBUG (MainThread) [pymodbus.logging] Processing: 0x1e 0x3 0x2 0x0 0x23 0x6c 0x5f
DEBUG (MainThread) [pymodbus.logging] Getting Frame - 0x3 0x2 0x0 0x23
DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
DEBUG (MainThread) [pymodbus.logging] Getting transaction 30
ERROR (MainThread) [custom_components.nilan.device] Could not read get_machine_type
ERROR (MainThread) [custom_components.nilan.device] Could not read get_bus_version
INFO (SyncWorker_11) [homeassistant.components.modbus.modbus] modbus Nilan communication open
```

master branch with replacement of method-name (async_pymodbus_call -> async_pb_call) worked though

veista, 2023-09-08:
Well that seems fairly odd since there should be no difference (I get this error too almost always, sometimes it works for a minute). From manifest.json could you change the minimum requirement for pymodbus to 2.4.1 and try again?

kfroeschl, 2023-09-08:
wasn't the issue but rather:

```diff
diff --git a/custom_components/nilan/device.py b/custom_components/nilan/device.py
index 95a28fd..c8d6946 100644
--- a/custom_components/nilan/device.py
+++ b/custom_components/nilan/device.py
@@ -38,7 +38,7 @@ class Device:
             "name": self._device_name,
             "type": self._com_type,
             "method": "rtu",
-            "delay": 1,
+            "delay": 0,
             "port": self._host_port,
             "timeout": 1,
             "close_comm_on_error": "false",
```

veista, 2023-09-08:
> wasn't the issue but rather:
>
> ```diff
> diff --git a/custom_components/nilan/device.py b/custom_components/nilan/device.py
> index 95a28fd..c8d6946 100644
> --- a/custom_components/nilan/device.py
> +++ [...]

That is funny. On my setup I cannot even connect with delay 0 on 2023.9

kfroeschl, 2023-09-09:
HA 2023.9.1 has updated pymodbus to 3.5.1

have you tried that already?
let me know if i should help testing.

jamespreedy, 2023-09-09:
Yeah although I'm not smart enough to help fix anything I did update to 2023.9.1 last night - both this Nilan modbus integration and the one I use for my Solax inverter are still having issues. 😥

veista, 2023-09-09:
> HA 2023.9.1 has updated pymodbus to 3.5.1
> 
> have you tried that already? let me know if i should help testing.

Noticed this morning that they pushed a new update. Will try today... I'm leaving on a trip again tomorrow so lets hope it works. Will anyway push a beta release so if it gets fixed on HA you can get this integration working again.

veista, 2023-09-09:
> Yeah although I'm not smart enough to help fix anything I did update to 2023.9.1 last night - both this Nilan modbus integration and the one I use for my Solax inverter are still having issues. 😥

I had to also go back to 2023.8.4 since the new update broke also z-wave and other stuff... I recommend going back and skipping this months release. I suspect it will take time untill all of the problems get fixed.

veista, 2023-09-09:
Also I suspect the issue #76 already has symptoms of something not working right with modbus. Though it has nothing to do with the root cause of that.

veista, 2023-09-09:
https://github.com/home-assistant/core/issues/99784#issuecomment-1712434631

veista, 2023-09-09:
> HA 2023.9.1 has updated pymodbus to 3.5.1
> 
> have you tried that already? let me know if i should help testing.

Tried the latest versions. Still the issues remain unfortunately. Released the partial fix as beta and reverted delay to 0.

gllmlbrt, 2023-09-13:
Hi, 
Updated to freshly released 2023.09.2, and with 1.2.5B of integration it all works perfectly now. All entities as normal.

gllmlbrt, 2023-09-13:
REsolved with 2023.09.2 and 1.2.5B

veikkoruuskanen, 2023-09-22:
Having a similar issue. Updated to 1.2.5B but still "Failed to connect" when I try to re-install the integration. To the log it gives the following error:

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/nilan/config_flow.py:56
Integration: Nilan (documentation, issues)
First occurred: 10:45:08 PM (3 occurrences)
Last logged: 10:47:46 PM

Failed to connect [Errno 111] Connect call failed ('192.168.0.111', 502)

veista, 2023-09-22:
> Having a similar issue. Updated to 1.2.5B but still "Failed to connect" when I try to re-install the integration. To the log it gives the following error:
> 
> This error originated from a custom integration.
> 
> Logger: pymodbus.logging Source: custom_components/nilan/config_flow.py:56 Integration: Nilan (documentation, issues) First occurred: 10:45:08 PM (3 occurrences) Last logged: 10:47:46 PM
> 
> Failed to connect [Errno 111] Connect call failed ('192.168.0.111', 502)
> 
> I restored an older full backup, and the integration works ok again on that so I assume that the bridge device is working OK. I have a RP4 and usb to rs485 adapter.

Had a similar issue on my dev environment. Could not replicate once is magically started working. Try the new release 1.2.6. If the issue remains just keep trying to install it again. The issue is somehow in the config flow, but cannot pinpoint the cause.

veikkoruuskanen, 2023-09-23:
Bumped to 1.2.6 with a HA restart after installing the HACS module. Still unable to install the integration:

Logger: pymodbus.logging
Source: components/modbus/modbus.py:396
First occurred: 12:01:22 PM (269 occurrences)
Last logged: 12:10:07 PM

Connection to (192.168.0.111, 502) failed: [Errno 111] Connection refused

veista, 2023-09-23:
It seems to me that for some reason the usb adaptor is busy. I will try and simulate this issue in my dev environment when I have time. Have you tried to do a Full reboot of your system? Power off and power on?

veista, 2023-09-23:
Ok. So I think I misunderstood your setup. You have made your own bridge device with a RPi and an USB adaptor?

The newer pymodbus is at the moment much less forgiving than the old one, so it might be an issue in both ends. I'll read the documentation again and try to figure out if I'm missing something in the update.

veikkoruuskanen, 2023-09-23:
Just to inform, the setup with 1.2.6 works ok now. I just restarted all, including the bridge device also. Did not need to manually touch anything, and the integration is now up and running ok. Thanks for the great piece of code 💪🏻

closed 2023-09-23 by veista