veista/nilan #105: Add support for VGU180 EK

jyrkij, 2023-12-04:
Hi,

Would it be possible to add support for VGU180 EK?  The error in debug log is:

```
2023-12-04 13:47:17.881 DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 38 not found in supported devices list
```

Here's the device type plate:

[image]

And I suppose this is the device info page:

[image]

veista, 2023-12-05:
Sure thing. I will add this ASAP.

veista, 2023-12-14:
Should be working in the latest release.

jyrkij, 2023-12-15:
Thank you!  The integration works, the device is recognised and communicates correctly.  There are missing registries though (and possibly some extra registries as well).  [2021-01-Modbus_CTS602_w_HMI350T_Modbus.pdf](https://www.en.nilan.dk/Files/Files/Engelsk/Downloads/7.%20Modbus%20-%20BACnet/2021-01-Modbus_CTS602_w_HMI350T_Modbus.pdf) lists at least central heating registries as supported (registry group 1800).  I haven't gone through the whole document so there might be some other registries missing as well.

veista, 2023-12-15:
As I don't have access to your device:
1. Add your device number to the regisry_map.py on the registries you think your device supports
2. Cross-check from your HMI panel that the registries actually work as intended
3. Create a pull request so that everybody can benefit from your work

Thanks!

PR #108 (2024-01-12): Set VGU180 EK supported registers
- VGU180 EK supports central heating
- supply air isn't physically present on this device
- humidity sensor isn't connected on my device so I cannot test that

Related to #107.  Resolves #105.
closed 2024-01-12 by veista