veista/nilan #110: Add support for Nilan CompactS

convicte, 2023-12-31:
Hi,

I was able to modify the `device_map.py file to include all the registers working for this device, and thus far it's working just fine.
There are some idiosyncrasies in the sensor names (e.g. T7 should be T2), but that is very minor.

Here is the output of the debug error:
`2023-12-30 23:47:17.881 DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 10 not found in supported devices list

I am not able to see the nameplate to provide a picture of it, but please see the requested Device Type information and the picture of the unit itself.

https://www.en.nilan.dk/products/ventilation-with-heating/ventilation-and-hot-water-production/compact-s

I'll follow up this initial report with a version of the `device_map.py` which works in my instance.

convicte, 2023-12-31:
[image]

[image]

[image]

veista, 2024-01-05:
Great! I will check your work asap. As for T2 -T7. T7 and T2 are the same sensor. T2 just stops working if you have an after heating element. I do not know why this is. T7 works always and T2 not. That is why just read T7 instead of trying to figure if someone has an after heating element.

veista, 2024-01-05:
And please do make a pull request with your changes.

closed 2024-02-04 by veista