veista/nilan #122: Latest update created orphaned device

convicte, 2024-02-19:
[image]

No way to delete it from HA without having to delete the integration, which I am afraid will mess all the sensors up as well.

veista, 2024-02-19:
I'm not sure what your issue is. This is something you have made yourself?

Just keep using your fork if you don't want to use this one.

convicte, 2024-02-19:
Thanks for the prompt reply.

I am not using a fork, but your repo all along since last year.

I've updated today to the latest version, which created a new device under the integration while the old device was left with 2 orphaned entities. Deleting these entities didn't help, and I am stuck with an empty device.

Not sure if deleting the integration all together to add it again may mess up the existing sensors any further.

Any ideas?

veista, 2024-02-19:
It would be nice to know which version number you are using.

convicte, 2024-02-19:
1.2.12 at the moment, updating from 1.2.10

veista, 2024-02-19:
#110 this is an issue you opened... support for CompactS was introduced in 1.2.12. I cant fix your code. Just reinstall or keep using your own fork.

closed 2024-02-19 by veista

convicte, 2024-02-19:
To reiterate, I was not using a fork, but can understand why you would think that.
It's also understandable that you get frustrated with the situation, since it's probably related to home assistant handling more than your integration.

I've ended up deleting the whole integration and re-downloading it. This deleted both devices without messing up the sensors.

I'll review the sensors in more detail as there are several climate entities which are not sensible. I'll try to create a pull request to correct this.

For example there is a separate climate entity for top and bottom water heater part even though there is no control over them separately. There is only one.

There are also some sensor duplicated, probably from another device.