veista/nilan #123: Can't initialize integration after 2024.4

veista, 2024-04-01:
Integration will not load.

jamespreedy, 2024-04-03:
Same, please let me know if I can provide any logs etc that could help. Thanks!

veista, 2024-04-03:
Sorry, I don't have time to look into this right now. Could not fix it easily, there is some issue introduced to the modbus wrapper on HA after making it async.

I would hold on updating/roll back for now.

AdamGrumlik, 2024-04-04:
2024.4.0 Integration not working:(

jamespreedy, 2024-04-04:
@AdamGrumlik - see viesta's comment above.
He's the author of the integration, is aware of the issue with 2024.4 but doesn't have time to look at this at the moment.

Best advise at this point is to roll back for now if you've already updated and the Nilan integration is required. 👍🏻

kfroeschl, 2024-04-04:
I think I've found the issue,

HA ModbusHub is now fully async:
https://github.com/home-assistant/core/commit/7cba34b2e6b53621e6188ff2b473a3553e72bdcd

and thus the `success = await self._modbus.async_setup()`([device.py](https://github.com/veista/nilan/blob/master/custom_components/nilan/device.py#L59)) call does no longer wait for the connection to be established.

in my local workaround i've added a `await asyncio.sleep(10)` right after to get it running again for now.
Though this is far from a beautiful solution it works for now.

(+ I've added the await before `await self._modbus.async_close()` to avoid another error)

veista, 2024-04-04:
Great! Can you make a pull request?

kfroeschl, 2024-04-04:
https://github.com/veista/nilan/pull/124

PR #125 (2024-04-06): Dev
Resolves #123 and updates #115
closed 2024-04-06 by veista