veista/nilan #127: Could not read errors - poss since 1.2.14

jamespreedy, 2024-04-13:
Hiya! Hope you're well.

I've recently been getting a bunch of errors in my HA logs - all in the same format.
Could not read get_alarm_1_code
Could not read get_pre_heater_defrost_select
etc etc

```
This error originated from a custom integration.

Logger: custom_components.nilan.device
Source: custom_components/nilan/device.py:2431
integration: Nilan (documentation, issues)
First occurred: 3:16:30 PM (9 occurrences)
Last logged: 3:39:32 PM

Could not read get_alarm_1_code
```

Also have ...
```
Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:318
integration: Modbus (documentation, issues)
First occurred: 3:37:30 PM (2 occurrences)
Last logged: 3:39:32 PM

Pymodbus: Nilan: Error: device: 30 address: 4104 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Pymodbus: Nilan: Error: device: 30 address: 1204 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries
```
and
```
Logger: homeassistant.components.climate
Source: components/climate/__init__.py:367
integration: Climate (documentation, issues)
First occurred: 4:50:21 PM (1 occurrences)
Last logged: 4:50:21 PM

Entity None (<class 'custom_components.nilan.climate.NilanClimate'>) implements HVACMode(s): cool, heat, auto, off and therefore implicitly supports the turn_on/turn_off methods without setting the proper ClimateEntityFeature. Please create a bug report at https://github.com/veista/nilan/issues
```

Would the debug logs help, or any particular troubleshooting I can do?
I've rebooted the Waveshare RS485 to Eth adaptor and power cycled my Combi Polar 302 - no change that I can see.
I've included the Waveshare settings below too, but I've not changed those for as long as I can remember.

[image]

Thanks in advance!

veista, 2024-04-14:
Is your integration still working?

veista, 2024-04-14:
Those errors are  so scarce it is normal. I guess there is some change in the modbus wrapper that makes it more prone to miss some messages.

Entity None (<class 'custom_components.nilan.climate.NilanClimate'>) implements HVACMode(s): cool, heat, auto, off and therefore implicitly supports the turn_on/turn_off methods without setting the proper ClimateEntityFeature. Please create a bug report at https://github.com/veista/nilan/issues

I will get this last one fixed! Thanks.

veista, 2024-04-14:
I updated to latest version of HA. There are some new issues introduced. Has nothing to do with version 1.2.14.

veista, 2024-04-14:
The issues have been resolved, but not yet released, the problem is with HA again.

jamespreedy, 2024-04-23:
@veista - Hi mate. Just FYI - HASS 2024.4.4 is out, and I installed Nilan 1.2.15.
Feels like HASS broke something again ... 😢

The integration wouldn't start and I'm getting the below errors.
Reverting to 1.2.14 make it work again.

```
Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:318
integration: Modbus (documentation, issues)
First occurred: 9:21:26 PM (2 occurrences)
Last logged: 9:21:33 PM

Pymodbus: Nilan: Error: device: 30 address: 1000 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.1.68:502]
```

```
This error originated from a custom integration.

Logger: custom_components.nilan.device
Source: custom_components/nilan/device.py:204
integration: Nilan (documentation, issues)
First occurred: 9:21:26 PM (2 occurrences)
Last logged: 9:21:33 PM

Could not read get_machine_type
```

```
Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:383
integration: Modbus (documentation, issues)
First occurred: 9:21:26 PM (2 occurrences)
Last logged: 9:21:33 PM

modbus Nilan communication closed
```

etc.

Please let me know if debug logs would help?
Thanks as always. 🍺

gllmlbrt, 2024-04-24:
I think I have the same issue and posted at https://github.com/veista/nilan/issues/132 before closing to avoid duplicate. But perahps the log there could help.

gabrieledaltoe, 2024-04-24:
Same for me...
```
2024-04-25 01:34:29.409 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: Nilan: Error: device: 30 address: 1000 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 10.3.5.253:502]
2024-04-25 01:34:29.409 ERROR (MainThread) [custom_components.nilan.device] Could not read get_machine_type
2024-04-25 01:34:29.409 WARNING (MainThread) [homeassistant.components.modbus.modbus] modbus Nilan communication closed
2024-04-25 01:34:29.409 ERROR (MainThread) [custom_components.nilan.device] Register hw_type returned None
2024-04-25 01:34:29.411 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
```

veista, 2024-04-25:
Everybody always has the same issue. You can like the original comment to keep updated and let know you have the issue too.

PMPoulsen, 2024-04-27:
Adding some information to the 1.2.15, in my system it will not configure at all. reverted back to 1.2.14 and things are working as above, with missing elements. i am running HAOS 2024.4.4.

Takethis88, 2024-05-03:
1.2.14 does not work with HA 2024.5.0

jamespreedy, 2024-05-03:
@Takethis88 - interestingly mine works great - no issues or errors in the logs with 1.2.14 with 2024.5.
Might be worth opening a new issue with debug logs etc?

@veista - all my issues are resolved now. As you say sounds like it was HA not your integration causing the issues.
Closing - thanks again for all your time and support to the project.