veista/nilan #130: Elfin EW11

lempparisi, 2024-04-20:
Hi,

I have connected Elfin EW11 RS485 - Wifi device to my Nilan EC9.

I had to install Nilan integration manually by copying the files. HACS download gave me an error about the HA version, I didn't know how to fix it.

I'm trying to use TCP settings but I don't know the right port or the device ID?

I can find the device in my local network with IP address.
I can see that Serial Port has traffic.

I connected the device like so: (Wire colors are from my installation) (This cabel goes from Nilan EC9 to CTS602)
Red: +10vdc, connect here the EW11 +
Blue: GND, connect EW11 -
Yellow: +1Vdc, connect EW11 A
White: +3Vdc, connect EW11 B

veista, 2024-04-20:
1. Im not sure if your bridge device is supported.

2. The current version does not work yet. Install older version via HACS

veista, 2024-04-21:
Hi,

I have no experience on your bridge device, but with a quick google it is supposed to work.

[image]

This mode is what you need. Ports is what you want default 502, also Unit ID is what you set at both ends, default 30.