veista/nilan #136: Sensor name and behaving

jusubbi, 2024-05-16:
Hello,

First thanks for awesome integration!

I have Nilan VP 18cek with firmware 2.35a and modbus version 9.

I think "sensor.nilan_tuloveden_lampotila_t14" is spelled wrong. It should be "sensor.nilan_menoveden_lampotila_t14" at least in this device because t14 is for outgoing water according to the manual.

Second thing is "sensor.nilan_keskuslammityksen_tavoitelampotila" Looks like it doesn't care about outside temperature and doesn't follow the curve at all. It only have some static changes during changes of "sensor.nilan_ohjaustila". And when "ohjaustila" is cooling, it goes to 0'C. I don't think it should behave like that? In summer we still need some floor heating in shower / sauna or is it unnecessary?

[image]
[image]
[image]

veista, 2024-05-16:
Hi,

You can suggest a new name as a pull request.  Edit fi.json and test on your setup. I can take a look if it is only on your model or is it a typo. If it is only on your model, I might not accept it.

As for functionality, this is a question for Nilan, not me. I can not change how your device functions. If you think there is another register that should be used/added, reference the user manual and I will try to add/change it.

closed 2024-05-16 by veista

jusubbi, 2024-05-17:
Moi,

En ole koodari, enkä saanut tuota pull requestia toimimaan, kun olisi
pitänyt viitata johonkin tiettyyn kohtaan koodissa jne.

Siis tässä on suoraan nilanin ohjekirjasta kuva tuosta kohdasta, missä
mainitaan noista sensoreista. Pitäisi latailla nilanin sivuilta muidenkin
laitteiden ohjekirjat ja tarkistaa onko niissä tuo T14 myös menoveden
anturi (vai onko integraatiossasi eri laitteille eri anturit?).
Jokatapauksessa kuvan perusteella se on tässä laitteessa (joka on Nilan EC9
nimellä myynnissä ollut) anturi sijaitseen sen kattilahässäkän sisällä.

Olisi kätevää, että olisi myös tuloveden anturi, niin näkisi paljonko sinne
lattiaan saa menemään.

Mutta kiitos vielä kerran hienosta integraatiosta, ilman tätä en varmaan
olisi saanut toimimaan HA:ssa mitenkään.

Seuraavaksi yritän saada scriptejä / automaatioita toimimaan, tähän asti
mikään triggeri ei ole toiminut jostain syystä..

-Juha