veista/nilan #139: Failure to read various numbers of values

JakobPP, 2024-05-26:
Hello veista!

Comfort 300
HASS OS 12.2
Home assistant 2024.5.5
Nilan HACS 1.2.16
RS-485 Waveshare ETH (B)

After the whole "Homeassistant 2024-05-xx" broke something modbus related I have been unable to read alot of registers. Like the humidity fx. Others are doing just fine. The logs just show errors and retries. Notthing has changed in the GW config.

[home-assistant_2024-05-26T18-51-49.184Z.log](https://github.com/veista/nilan/files/15448599/home-assistant_2024-05-26T18-51-49.184Z.log)

veista, 2024-06-02:
Unable to read as in some registers/functions stopped working or logs just show various registers that could not be read at some point? It could be any reason why your logs have failed read attempts (mine have them too with various devices)

JakobPP, 2024-06-02:
Stopped working, is my best guess. But i honestly cant say.Venlig hilsen - - Best regardsJakob Pilgaard Pedersen

veista, 2024-06-02:
Ok? Does your humidy show values?

veista, 2024-10-31:
Closing this as stale.

closed 2024-10-31 by veista

JakobPP, 2024-11-24:
Hello again Veista

Apologies for stale.

But, this issue hasnt improved or disappeared. Im running lates HASS, HACS 2.x and the lastest of your Nilan app. And still integrating via the Waveshare.

I would still like to test out slowing down poll speed, but I cant find such a setting. I can read intake temperature, bypass flap state,  return air temperature etc. But humidity wont read, exhanger effiency as well and so on for alot if not most of the entities. In the process of upgrading to HACS 2.x I removed/deleted the Nilan device and integration via the HASS UI, and then installed again. Is there another way of making sure any old components of your integration is completely removed before trying a fresh install?

Any hope of getting your support on this? Attaching the debug log.

[Uploading home-assistant_2024-11-24T11-55-03.368Z.log…]()

JakobPP, 2025-04-13:
For anyone reaching this thread. The issue seems to have been as stupid and simple as this:

A hard reboot (power off/power on) of the Waveshare gateway fixed the issue.