veista/nilan #144: Operation mode is not valid - Valid operation modes are: Off, On. - CompactS

convicte, 2024-08-09:
Error displayed:
```Operation mode is not valid for water_heater.nilan_bottom_water_heater. Valid operation modes are: Off, On.```

I am on the latest version (1.2.16) and it seems like there is some strangeness going on with turning the water heater ON/OFF.
It's been like this for the longest of time, but only recently turned it OFF/ON to see the error.

Before I was on a local fork, which I didn't care to fix all stuff on, now having switched to the main repo in production the error is still there.

The modes in the UI are ON and OFF, but it seems to report they are no in the commands being sent, so there is probably some ON=true, OFF=false or similar situation
[image]

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:283
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 1:36:41 PM (4 occurrences)
Last logged: 1:37:16 PM

[546835333472] Operation mode is not valid for water_heater.nilan_bottom_water_heater. Valid operation modes are: Off, On
[546835333472] Operation mode is not valid for water_heater.nilan_top_water_heater. Valid operation modes are: Off, On
```

Not sure if this is related, as it floods logs as well:

```
Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:331
integration: Modbus (documentation, issues)
First occurred: 12:50:54 PM (412 occurrences)
Last logged: 1:41:56 PM

Pymodbus: Nilan: Error: device: 30 address: 1500 -> pymodbus returned isError True
Pymodbus: Nilan: Error: device: 30 address: 4018 -> pymodbus returned isError True
Pymodbus: Nilan: Error: device: 30 address: 4027 -> pymodbus returned isError True
Pymodbus: Nilan: Error: device: 30 address: 1702 -> pymodbus returned isError True
```

veista, 2024-08-09:
I have the same issue. Thanks for reporting. I will take a look during the weekend.

convicte, 2024-08-09:
I am seeing a couple more popping up in my logs, but don't want to conflate them into one for the moment.
When this one is resolved, I'll most the remaining errors.

Thanks for looking into this!!

convicte, 2024-09-30:
> I have the same issue. Thanks for reporting. I will take a look during the weekend.
Sorry to be a bother - and chance you will find a moment to have a look?

Cheers!!

veista, 2024-09-30:
I will find some time this week. Thank you for your patience.

closed 2024-10-30 by veista

convicte, 2024-10-30:
Installed and tentatively tested to not throw and error.
Will report if anything pops up - thank you!