veista/nilan #149: Unsupported Device VP18C Compact M2 M/køl

DUC750, 2024-09-30:
Could it be possible to integrate my Nilan device? 
I hope I have included all the necessary data. 
Thank you in advance.

[image]

[image]

Log: 2024-09-30 20:58:53.751 DEBUG (MainThread) [custom_components.nilan.config_flow] Device Type 11 not found in supported devices list

veista, 2024-09-30:
Sure thing. I will add it this week.

DUC750, 2024-09-30:
Übersetze ins englische: so cool! Ich freue mich tierisch darauf. Vielen vielen Dank 😃

DUC750, 2024-09-30:
So cool! I’m really looking forward to it. Thank you so much 😃

DUC750, 2024-10-14:
Hi @veista, I wanted to ask if you have already added the device? 
Regards, DUC

veista, 2024-10-14:
Hi, I am terribly sorry, but I have simply no time right now. I will add this as soon as I can. Thank you for your patience and understanding.

DUC750, 2024-10-14:
Oh no, that’s not what I meant. I have no intention of rushing someone who develops such great things for the community in their free time. I just wanted to ask if I missed something. It takes as long as it takes. :-)

DUC750, 2024-10-30:
Hello @veista,

I was a bit impatient and went ahead with the implementation of my device myself. Everything seems to be working fine so far. If you like, you can use the attached file.
[device_map.zip](https://github.com/user-attachments/files/17573497/device_map.zip)

Cheers,
DUC

veista, 2024-10-30:
That is great. If you could do a pull request, that would be perfect.

DUC750, 2024-10-30:
Never done it before, but there’s always a first time for everything. I’ll give it a try :-)

DUC750, 2024-10-30:
#151

closed 2024-10-30 by DUC750