veista/nilan #152: Get_user_function_2_state - set to supported None while most likely should be all

convicte, 2024-10-30:
This feature is the second relay switch available on many (if not all) Nilan units, to allow triggering from external devices, such as kitchen hoods and similar
``` yaml
    "get_user_function_2_state": {
        "entity_type": "binary_sensor",
        "min_bus_version": 2,
        "min_hps_bus_version": 1,
        "supported_devices": (None,),
```
It's currently listed as supported by `NONE` but I would recommend switching to `all`, which in a worse case scenario would be disabled as sensor by user.

veista, 2024-10-30:
Hi,

Good to know, you are probably correct. I will try to add this feature in the next release.

convicte, 2025-01-05:
Just friendly reminder, since this must have slipped through in the latest release.

Every time I update it breaks my locally edited copy for the hood extractor connected to the unit.

Thanks in advance!

veista, 2025-01-12:
Did not slip my mind. I want the entity to be disabled by default and it need a bit more implementation.

closed 2025-02-23 by veista

convicte, 2025-02-23:
> What is an unfriendly reminder?
Someone being rude and or demanding of a maintainer for missing something they'd wish to get.
People are amazingly bad behaved when it comes to these issues on the web.