veista/nilan #157: Could not read get_display_text_1 + 2

ThomasBLarsen, 2024-11-08:
I have adapted the code to suit a Nilan Compact_N (Nordic compact P model) and I see that the code tries to read the LED state for all modells. I have the CTS602 with HMI Touch panel which does not have any led, and apparently it seems like the modbus addresses are removed for my firmware.  Does every other version of the panel have the LED states? I see that the get_display_led_1_state is has "supported_devices": ("all",). If this is the case, can we maybe make a boolean in the config flow that ask the user if they have the HMI interface or not?

Logger: custom_components.nilan.device
source: custom_components/nilan/device.py:334
integration: Nilan (dokumentasjon, problemer)
first seen 11:30:26 (161 hendelser)
Last logged: 12:50:27

Could not read get_display_text_2

FIY. I will send a PR with the Compact_N profile as soon as get it stable on my HA.

veista, 2024-11-08:
Hi,

Sounds great. You are mistaken. The functionality is primarly determined by the modbus version (protocol version). Which is different depending on display etc. I think your model is another edge case, where you need to adhere to the compact +air/geo definitions. This is defined in device map, but also you need to add this device number to the device handling logic.

ThomasBLarsen, 2024-11-08:
```
    "get_display_led_1_state": {
        "entity_type": "binary_sensor",
        "min_bus_version": 1,
        "max_bus_version": 19,
        "supported_devices": ("all",),
    },
    "get_display_led_2_state": {
        "entity_type": "binary_sensor",
        "min_bus_version": 1,
        "max_bus_version": 19,
        "supported_devices": ("all",),
    },
    "get_display_text_1": {
        "entity_type": "sensor",
        "min_bus_version": 1,
        "max_bus_version": 19,
        "supported_devices": ("all",),
    },
    "get_display_text_2": {
        "entity_type": "sensor",
        "min_bus_version": 1,
        "max_bus_version": 19,
        "supported_devices": ("all",),
```

These entities are the ones failing for my device, but the code says "supporte_devices" : ("all",),. Would the right way to handle this be to replace "all" with all of the device ID's execept for the model I am running (30 in the case for COMPACT_N)?

veista, 2024-11-08:
No. It is not that simple.

veista, 2024-11-08:
If you do that you can fix it for you, but you will not be able to receive updates in the future. I will not accept such thing in this integration, since it would break everything. You need to understand the code to be able to solve this on your own.

ThomasBLarsen, 2024-11-08:
Sorry, I did not intending to offend, I just wanted to make sure I understood you correctly. 

I written my own HA-integration before I discovered your integration which was more generic and mature. I thought I would help out as I also have access to the Nilan firmware files and have work quite a lot with the Modbus. 

What is the best way to resolve this? Should I start looking into the modbus to find out what distinguishes the HMI interface from the classic one?

veista, 2024-11-08:
Can you share your modbus version, device id, and the pictures asked in the readme? As I said the logic is already implemented, but the device ID has to be added to it. Without this information I cannot help at all.

ThomasBLarsen, 2024-11-09:
Device ID is 30, I will post the Modbus version when I find it. 

[image]

veista, 2024-11-09:
Hi,

Ok.

Are you sure it is 30. Did you get this from the debug log or where?

ThomasBLarsen, 2024-11-09:
I got it from debugging the Nilan component and and it also corresponds to the ID I have extracted before.

veista, 2024-11-18:
Have you found the Bus Version?

ThomasBLarsen, 2024-11-18:
Bus version is 13. I see that you also check for a hps_bus_version, but I can't see that printed in the logs?

ThomasBLarsen, 2024-11-18:
I also have also been working on mapping up the device, but I am not done yet.

veista, 2024-11-18:
Thanks. Unfortunately this means Nilan has made another outlier of this device.

Have you asked Nilan for a protocol manual for this? Could you please do this and provide that to me?

ThomasBLarsen, 2024-11-29:
I made a table of all the firmwares available from the Nilan aftermarket site. Maybe the firmware version number is better for determining the capabilites?

| **Model**          | **File Name**                                         | **Date**            | **Size**   | **Notes**       |
|---------------------|------------------------------------------------------|---------------------|------------|-----------------|
| **CTS 602**        | CTS-602_2.17a.bin                                     | 04-08-2020 08:45:35 | 212 KB     |                 |
|                     | CTS-602_2.17e.bin                                     | 04-08-2020 08:45:35 | 213 KB     |                 |
|                     | CTS-602_2.17g.bin                                     | 04-08-2020 08:45:35 | 213 KB     |                 |
|                     | CTS-602_2.21a.bin                                     | 04-08-2020 08:45:35 | 215 KB     |                 |
|                     | CTS-602_2.24a.bin                                     | 04-08-2020 08:45:35 | 256 KB     |                 |
|                     | CTS-602_2.30e.bin                                     | 04-08-2020 08:45:35 | 261 KB     |                 |
|                     | CTS-602_2.31d.bin                                     | 04-08-2020 08:45:35 | 263 KB     |                 |
|                     | CTS-602_2.33c.bin                                     | 04-08-2020 08:45:35 | 266 KB     |                 |
|                     | CTS-602_2.34c.bin                                     | 04-08-2020 08:45:35 | 266 KB     |                 |
|                     | CTS-602_2.35d.bin                                     | 04-08-2020 08:45:35 | 267 KB     |                 |
|                     | CTS-602_2.36a.bin                                     | 04-08-2020 08:45:35 | 268 KB     |                 |
|                     | CTS-602_2.37c.bin                                     | 04-08-2020 08:45:35 | 268 KB     |                 |
|                     | CTS-602_2.38.r_crc.bin                                | 25-02-2021 08:02:36 | 302 KB     |                 |
|                     | CTS-602_2.38.t_crc.bin                                | 13-01-2022 15:56:16 | 308 KB     |                 |
| **CTS 602 HMI**    | Pack_CTS_602_1.6.20.0_1.6.20.0.bin                     | 04-08-2020 08:46:49 | 692 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.21.0_1.6.21.0.bin                    | 04-08-2020 08:46:49 | 710 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.22.0_1.6.22.0.bin                    | 04-08-2020 08:46:49 | 715 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.24.0_1.6.24.0.bin                    | 04-08-2020 08:46:50 | 717 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.26.0_1.6.26.0.bin                    | 04-08-2020 08:46:50 | 717 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.40.0_1.6.40.0.bin                    | 04-08-2020 08:46:50 | 721 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.42.0_1.6.42.0.bin                    | 02-09-2020 13:25:09 | 722 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.45.0_1.6.45.0.bin                    | 25-02-2021 08:01:50 | 725 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.48.0_1.6.48.0.bin                    | 26-05-2021 14:17:27 | 756 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.49.0_1.6.49.0.bin                    | 25-06-2021 08:43:37 | 757 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.54.0_1.6.54.0.bin                    | 21-09-2021 14:49:37 | 771 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.56.0_1.6.56.0.bin                    | 26-10-2021 15:18:38 | 778 KB     | OLD CTS602      |
|                     | Pack_CTS_602_1.6.57.0_1.6.57.0.bin                    | 26-04-2022 11:58:13 | 872 KB     | OLD CTS602      |
|                     | Pack_CTS_602_3.6.57.0_HMI350T_3.6.57.0.bin            | 12-10-2022 11:11:55 | 883 KB     | New CTS602      |
|                     | Pack_CTS_602_1.6.60.0_HMI350T_3.6.60.0.bin            | 15-04-2024 12:05:10 | 1 KB       | OLD CTS602      |
|                     | Pack_CTS_602_3.6.60.0_HMI350T_3.6.60.0.bin            | 15-04-2024 12:05:17 | 1 KB       | New CTS602      |
|                     | Pack_CTS_602_1.6.61.0_HMI350T_3.6.61.0.bin            | 15-04-2024 12:00:45 | 1 KB       | OLD CTS602      |
|                     | Pack_CTS_602_3.6.61.0_HMI350T_3.6.61.0.bin            | 15-04-2024 12:00:53 | 1 KB       | New CTS602      |
|                     | Pack_CTS_602_1.6.64.0_HMI350T_3.6.64.0.bin            | 17-04-2024 09:15:18 | 905 KB     | OLD CTS602      |
|                     | Pack_CTS_602_3.6.64.0_HMI350T_3.6.64.0.bin            | 17-04-2024 09:15:22 | 906 KB     | New CTS602      |
| **CTS 602 HMI - AIR** | AIRPack_1.1.6.0_Air_1.1.6.0_Hmi_1.1.6.0_Remote_1.1.6.0.bin | -                   | -          | OLD CTS602      |
|                     | AIRPack_4.1.6.36_Air_4.1.6.36_Hmi_1.1.6.36_Remote_1.1.6.36.zip | -                 | -          | New CTS602      |
| **CTS 602 HMI - GEO** | GEOPack_1.2.6.0_Geo_1.2.6.0_Hmi_1.1.6.0_Remote_1.1.6.0.bin | -                  | -          | OLD CTS602      |
|                     | GEOPack_4.2.6.36_Geo_4.2.6.36_Hmi_1.1.6.36_Remote_1.1.6.36.zip | -                | -          | New CTS602      |
[... cut ...]
|                     | Pack_CTS_602_L_1.1.25.0_1.1.23.0.bin                  | 04-08-2020 08:47:01 | 669 KB     | OLD CTS602      |
|                     | Pack_CTS_602_L_1.1.28.0_1.1.28.0.bin                  | 04-08-2020 08:47:01 | 676 KB     | OLD CTS602      |
|                     | Pack_CTS_602_L_1.1.29.0_1.1.29.0.bin                  | 25-06-2021 08:57:55 | 716 KB     | OLD CTS602      |
|                     | Pack_CTS_602_L_1.1.30.0_1.1.30.0.bin                  | 17-09-2021 09:10:48 | 726 KB     | OLD CTS602      |

veista, 2024-12-01:
Did you ask Nilan for the manual?

veista, 2025-01-12:
Do you have any update to this?

closed 2025-02-22 by veista