veista/nilan #160: Nilan Compact P (air/geo)  GEO entities missing from the list.

Hannes165, 2024-11-30:
Hi,

Added waveshare rs485/wifi adapter to the nilan compact P machine (replaced the original wifi gateway). Configured everything successfully and got the connection established.
In HA, my nilan shows up with 50 entities. Most of the GEO part (device 44)parameters seems to be missing. Any idea how to get them visible? Or what to check for troubleshooting?

veista, 2024-12-01:
What specifically is missing? Can you share the requested information?

Hannes165, 2024-12-01:
Added the pictures of available entities and device info.
[image]
[image]
AIR side seems to be all there. CO2 value didnt find.
GEO side. Brine (geo circuit) temp in out missing. Floor circuit in out missing. Compressor state and load missing. Pressure data missing. These should all be in modbus addresses 200-222. T14 is indicating the supply water temp, and it actually is this temp, but on the mashine itself, this temperature is marked  T17. All setpoints missing.

Modbustester tool gives me all values if testing adresses regarding the registry from the nilan HACK files. This indicates that the connection details should be OK.

I can provide more info, i just do not know what specific info needed😉

Hannes165, 2024-12-05:
Hi,

 

Could you help me on the Nilan topic?

 

Tervitused / Best regards

Hannes Toomsalu

veista, 2024-12-06:
Turn debug logging on and reinitialize the integration. Give me the log.

Hannes165, 2024-12-06:
Hi,

 

Enabled Debug Logging and reloaded the integration. Here is the log file.

 

Please note. My nilan Unit is running on latest software and the original MODBUS extender is disconnected. Waveshare is connected straight to the unit.

 

Tervitused / Best regards

Hannes Toomsalu

veista, 2024-12-06:
You need to reply on the website.

Hannes165, 2024-12-06:
[home-assistant_nilan_2024-12-06T13-42-24.877Z.log](https://github.com/user-attachments/files/18039020/home-assistant_nilan_2024-12-06T13-42-24.877Z.log)

veista, 2024-12-06:
Could you please filter the log to only include nilan integration.

veista, 2024-12-06:
0. Enable debug logging for integration
1. Clear logs
3. Reload nilan integration
4. Search custom_components.nilan in logs
5. View raw log
6. Enable condensed view

[image]

Hannes165, 2024-12-06:
is that OK?
[home-assistant_2024-12-06T16-29-02.525Z.log](https://github.com/user-attachments/files/18040961/home-assistant_2024-12-06T16-29-02.525Z.log)

veista, 2024-12-06:
> is that OK? [home-assistant_2024-12-06T16-29-02.525Z.log](https://github.com/user-attachments/files/18040961/home-assistant_2024-12-06T16-29-02.525Z.log)

Does this look like the example I gave you?

Hannes165, 2024-12-06:
[home-assistant_2024-12-06T17-35-14.110Z.log](https://github.com/user-attachments/files/18041611/home-assistant_2024-12-06T17-35-14.110Z.log)

Hannes165, 2024-12-06:
settings/system/logs...
[image]
Hopefully this will help😉

veista, 2024-12-06:
Yes. So you have a model with AIR or GEO installer? According to the integration logic you do not.

veista, 2024-12-06:
> Added the pictures of available entities and device info. ![available after installation](https://private-user-images.githubusercontent.com/115191457/391347228-0384e65c-bb07-437d-9856-7be341afafae.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1MDczNDAsIm5iZiI6MTczMzUwNzA0MCwicGF0aCI6Ii8xMTUxOTE0NTcvMzkxMzQ3MjI4LTAzODRlNjVjLWJiMDctNDM3ZC05ODU2LTdiZTM0MWFmYWZhZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwNlQxNzQ0MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMDEyZWFhNGIwMzJjY2Y1ZWZjMTg5MWMwZDYzNmQwMjk5ZTM3NjY0N2NlYWIxMTE0YjhjNjU5OGMyMmRjNzY1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ubt2tliYQ54lUCAYxn-pQNAWq-z86oJjTm6WIC5OaWw) ![device info after installation](https://private-user-images.githubusercontent.com/115191457/391347231-4187a5e1-ca90-4050-a968-7e88b8d1b70c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1MDczNDAsIm5iZiI6MTczMzUwNzA0MCwicGF0aCI6Ii8xMTUxOTE0NTcvMzkxMzQ3MjMxLTQxODdhNWUxLWNhOTAtNDA1MC1hOTY4LTdlODhiOGQxYjcwYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwNlQxNzQ0MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMTg5ODM4NTE1MWFlZjY0ZDQzYWY3OTYzOTczNjc2YzA5ZGMzOWNmMWFkYTlkYjk4MDdhOWRjMjNmN2ViOTQ2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Nz5QAeCmsUqyKJml1IU7Evpie_HkaWA1ofJnc5ixCYM) AIR side seems to be all there. CO2 value didnt find. GEO side. Brine (geo circuit) temp in out missing. Floor circuit in out missing. Compressor state and load missing. Pressure data missing. These should all be in modbus addresses 200-222. T14 is indicating the supply water temp, and it actually is this temp, but on the mashine itself, this temperature is marked T17. All setpoints missing.
> 
> Modbustester tool gives me all values if testing adresses regarding the registry from the nilan HACK files. This indicates that the connection details should be OK.
> 
> I can provide more info, i just do not know what specific info needed😉

Also you can't have both AIR and GEO installed, so you must be misunderstanding something here.

veista, 2024-12-06:
Take a picture of the device type plate and submit it with the issue.
If you have HMI350T - Touch screen HMI - installed on your device, take a picture of the device info page and submit it with the issue.
If you have CTS602 HMI, take a picture of "SHOW DATA" -> "TYPE" and submit it with the issue.

Can you provide these. Thanks!

Hannes165, 2024-12-06:
device Type plate:
[image]
CTS602 HMI/show data/unit information:
[image]
[image]
CTS602 board:
[image]

veista, 2024-12-06:
Thanks. There is obviously a problem with the GEO recognition logic.

Hannes165, 2024-12-06:
I have installed the CO2 sensor and i can read it from HMI, also can read it with modbustester. The unit has two heat pumps inside, one for AIR, second for GEO.
HMI home screen:
[image]
Show data has two units under it:
[image]
Geo unit data:
[image]
[image]
[image]
[image]
[image]
This and the AIR side sensors are under modbus addresses 200-222.

veista, 2024-12-06:
AIR is an external heatpump, so you misunderstand that functionality.

Hannes165, 2024-12-06:
[image]
[image]

veista, 2024-12-06:
> OK. In my case i was refering the air side heat pump inside my nilan device. It has Ventilation unit with heat pump on the extraction side. Heating supply AIR and DHW. Second part on the bottom of the device is GEO thermal heat pump primerily for floor heating, also available for DHW heating if system built acordingly.

I see, that is just confusing! AIR is air-water heatpump and you are talking about the regular indoor heatpump. All of this should be implemented, you are probably the first user to have this device. If you can manually read thf register, we can debug this easily.

Hannes165, 2024-12-06:
I will add here the Show Data/Compact P as well:
[image]
[image]
[image]
[image]
[image]
[image]

veista, 2024-12-06:
I think I found the issue. It will take some time to test and implement.

Hannes165, 2024-12-06:
[image]

veista, 2024-12-06:
Most issues should be resolved with latest release!

Hannes165, 2024-12-07:
update installed, HA restarted. It is impressive. Now there are two devices visible.  Most of the sensors are on GEO side.  For the Air unit compressor the state is running, but for the geo unit compressor there is no state indicated. Compressor voltage and capacity is visible but no state indication. Shall i try to list the sensors whitch should be on the AIR side of the nilan unit? Circulation pump is a mysterious thing. The geo unit has circulation pump for the Brine circuit, and this is Off right now. On top of that there is floor heating pump what is always running if in winter mode. And this is connected to one of the relays on the board. This can not be seen from the list. Otherwise the number of entities is now 116.

[image]

[image]

Hannes165, 2024-12-07:
There are some minor things like LED 1; 2. Display row 1; 2. These are not for device 44 i think. Also T2 is no longer visible. Supply air temp is visible, but with T7 on its name. On device 44 it is called Supply air T2.
Also the status of Brine circulation pump, unknown. And the status of the Floor heating circulation pump. Can i somhow assist you to get this added/fixed

Hannes165, 2024-12-07:
There are some minor things like LED 1; 2. Display row 1; 2. These are not for device 44 i think. Also T2 is no longer visible. Supply air temp is visible, but with T7 on its name. On device 44 it is called Supply air T2.
Also the status of Brine circulation pump, unknown. And the status of the Floor heating circulation pump. Can i somhow assist you to get this added/fixed.

Water setpoint is DHW heating setpiont for GEO pump, if enabled. Heating setpoint is the temperature the pump wants to supply to the heating (T17) and this is dynamically chande acording to the outdoor temperature temperature compensation curve.
Here are te GEO side sensors.
[image]

Hannes165, 2024-12-07:
I will leave tomorrow for some weeks. I can not give any feedback during that time. Many thanks and will revert later if anything comes up.

veista, 2024-12-07:
> There are some minor things like LED 1; 2. Display row 1; 2. These are not for device 44 i think. Also T2 is no longer visible. Supply air temp is visible, but with T7 on its name. On device 44 it [...]

Yes. Remove integration. Add integration.

veista, 2024-12-07:
> There are some minor things like LED 1; 2. Display row 1; 2. These are not for device 44 i think. Also T2 is no longer visible. Supply air temp is visible, but with T7 on its name. On device 44 it is called Supply air T2.
> Also the status of Brine circulation pump, unknown. And the status of the Floor heating circulation pump. Can i somhow assist you to get this added/fixed.
> 
> Water setpoint is DHW heating setpiont for GEO pump, if enabled. Heating setpoint is the temperature the pump wants to supply to the heating (T17) and this is dynamically chande acording to the outdoor temperature temperature compensation curve.
> Here are te GEO side sensors.
> [image]
> 
> 
> I will take a look. Most likely you will have to rename these yourself.

veista, 2024-12-07:
I will take a look at wrongly named entities. This might not be fixable at integration level. They might be different only on your device. I can't rename only on your device.

If entities are missing, please reference manual and HMI panel. I will take a look into adding these in the future.

veista, 2025-02-23:
I'm closing this as stale. Please feel free to open a new issue with any missing feature.

closed 2025-02-23 by veista

veista, 2025-03-10:
I'm so sorry, but I can't help you if you don't tell me what you want to achieve. Also I am very limited on time, so I am not able to hold your hand on this.

Hannes165, 2025-03-10:
Hei,

 

No problem. Let it be right now. It is already better than at the beginning.

👍

 

Tervitused / Best regards

Hannes Toomsalu