veista/nilan #167: Add Compact P Installation Guide to Wiki ("pull request")

bboto32, 2025-01-09:
Hi, 

during my battle with installation, I really missed some guide that would have speed up the process so I prepared one. It's a new Wiki page [Compact P Installation Guide](https://github.com/bboto32/nilan/wiki/Compact-P-Installation-Guide) in the forked repo:
https://github.com/bboto32/nilan.wiki.git

Since I haven't found a better way, this is the "pull request" to add it to the original wiki.

Bohumir

veista, 2025-01-09:
Sorry. Apparently it is not possible to pull request on wiki on github. I will copy this with minor edits to the wiki. Thanks!

bboto32, 2025-01-09:
No problem. Thank you for doing all the important work!

veista, 2025-01-09:
I will try to make PR:s possible for wiki so you can make a PR to it yourself. I will keep you posted.

veista, 2025-01-12:
Can you edit the wiki now?

bboto32, 2025-01-13:
Yes, but I was able to add new page directly, without a PR.

veista, 2025-01-13:
Ok. That could be problematic.