veista/nilan #168: "Unknown error occured" when adding integration

jmaciak, 2025-01-12:
Hello!

I would like to report an issue which I am unable to resolve:

- HAOS
    - Core 2025.1.2
    - Supervisor 2024.12.3
    - OS 14.1
    - Frontend 20250109.0
- Trying to connect to Nilan via Modbus TCP on port 502, communication is checked and verified via other 3rd party tools

1. Installed integration version 1.2.18 from HACS
2. Then trying to add integration
    - Select Interface Type: TCP
    - Device Name: Nilan
    - IP Address: 192.168.240.28
    - Port: 502
    - Unit ID: 30
3. After clicking on Submit, the following error occurs:

[image]

Additional information was found in Home Assistant Core logs:

```
Logger: aiohttp.server
Source: /usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py:451
First occurred: 12:24:00 (3 occurrences)
Last logged: 13:23:45

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 210, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
[... 119 lines cut ...]
  File "/config/custom_components/nilan/config_flow.py", line 106, in async_step_tcp
    await async_validate_device(
    ...<4 lines>...
    )
  File "/config/custom_components/nilan/config_flow.py", line 40, in async_validate_device
    client = AsyncModbusTcpClient(address, port)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/tcp.py", line 79, in __init__
    
    
    ...<4 lines>...
            "Connecting to {}:{}.",
            ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 41, in __init__
        **reconnect_delay** to **reconnect_delay_max**.
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    
    
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/modbusclientprotocol.py", line 38, in __init__
KeyError: '502'
```

Please any advice how to move forward? If I can provide any additional information/diagnostics, I will be happy to help.

Thank you for your help and your work!

veista, 2025-01-12:
Can you give me your routers IP and subnet mask. Also be sure that nothing else is connected to the bridge device.

jmaciak, 2025-01-12:
Dear Taneli,

I have rather complicated network topology with VLANs, but the VLAN where
both Home Assistant and the Waveshare RS485 TO POE ETH (B) are, is:

Router: 192.168.240.1
WaveShare: 192.168.240.28
HAOS: 192.168.240.10
Network: 192.168.240.0
Netmask: 255.255.255.0

I have a PC in the same VLAN, and I can connect to WaveShare using
application called ModbusTools on IP 192.168.240.28 and port 502, and
communicate with Nilan (I tried to read several input registers as
described in Nilan Modbus protocol for my unit and I have seen expected
values, therefore I can confirm the Modbus communication is working
properly). Before trying to add the integration, I indeed closed the
communication with WaveShare from my PC.

In the meantime, I have found this: 2025.1.0 modbus integration fails to
load · Issue #134593 · home-assistant/core [github.com]
<https://github.com/home-assistant/core/issues/134593>

It looks like Home Assistant 2025.1 breaks some dependencies with pymodbus
for other people, describing similar issues with pymodbus framer, however I
do not understand what that means.

Thank you!

Juraj

veista, 2025-01-12:
Okay. I'm not sure how can I help with this. I will try to replicate the issue, but if I can't, I'm afraid I don't have time to go deeper into this issue.

veista, 2025-01-12:
This should be fixed in latest release

jmaciak, 2025-01-12:
Aaaand it indeed is! Thank you so much! Everything is working as expected

closed 2025-01-12 by veista