veista/nilan #169: "Unknown error occured" when adding integration via Serial

vojtechalbl, 2025-01-16:
Hello,

after updating and rebooting my OpenMediaVault server I am unable to connect Nilan device to my HA instance:

HA is running in docker.

Core: 2025.1.2
Frontend 20250109.0

Hardware:

USB to RS485 CH340c

Installed integration version 1.2.19 from HACS

Then trying to add integration
Select Interface Type: 
Serial
Device Name: 
Nilan
Path:
/dev/ttyUSB0
Unit ID: 
30
After clicking on Submit, the following error occurs:

[image]

HA logs:

                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/nilan/config_flow.py", line 134, in async_step_serial
    await async_validate_device(
        "serial", user_input["host_port"], user_input["unit_id"], None
    )
  File "/config/custom_components/nilan/config_flow.py", line 46, in async_validate_device
    client = AsyncModbusSerialClient(
        method="rtu",
    ...<5 lines>...
        timeout=1,

I tried checked dev/ttyUSB twice, reinstalled integration, restarted HA several times. Device od /dev/ttyUSB1 works correctly. 

Hardware is connected to sever and docker can see from container. Just before server reboot it worked flawlessly.

Any suggestions how to solve this problem? 

Do you need any further information?

Thanks in advance for reply.

veista, 2025-01-16:
Needs an update. I did not catch all changes to pymodbus infortunately. I will fix this during the weekend.

vojtechalbl, 2025-01-30:
Any news, will you be able to fix it? Or the changes to pymodbus are so big, that only TCP connection will remain usable?

closed 2025-02-23 by veista