veista/nilan #171: After HA core 2025.2 update. Nilan inegration 1.2.19. Failed to perform the action climate/set_temperature. object of type 'int' has no len()

Almetu, 2025-02-06:
Home Assistant: 
Core 2025.2.0
Supervisor 2025.02.0
Operating System 14.2
Frontend 20250205.0

Nilan integration:
1.2.19

Device info
VP 18cek
by Nilan
Firmware: 1.6.64
Hardware: 16

Connection TCP/Modbus - Waveshare RS485 to RJ45 Ethernet (B) PoE

_****After update HA core to ver. 2025.2 there is impossible to perform and control actions and change values (Fan mode, heat/cooling mode, etc.) (write data via Modbus)**_ 

- Failed to perform the action climate/set_temperature. object of type 'int' has no len()
- Failed to perform the action climate/set_fan_mode. object of type 'int' has no len()

logs (debug):

1)
Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:310
integration: Modbus (documentation, issues)
First occurred: 11:50:07 (1 occurrences)
Last logged: 11:50:07

2)

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 11:39:11 (6 occurrences)
Last logged: 11:57:03

[140379875756928] Unexpected exception
[140379027960688] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/climate/__init__.py", line 955, in async_service_temperature_set
    await entity.async_set_temperature(**kwargs)
  File "/config/custom_components/nilan/climate.py", line 153, in async_set_temperature
    await self._device.set_user_temperature_setpoint(kwargs[ATTR_TEMPERATURE])
  File "/config/custom_components/nilan/device.py", line 3303, in set_user_temperature_setpoint
    await self._modbus.async_pb_call(
    ...<4 lines>...
    )
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 424, in async_pb_call
    result = await self.low_level_pb_call(unit, address, value, use_call)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 389, in low_level_pb_call
    result: ModbusPDU = await entry.func(address, **kwargs)
                              ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 514, in write_registers
    return self.execute(no_response_expected, pdu_reg.WriteMultipleRegistersRequest(address=address, registers=values,dev_id=slave))
                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/pdu.py", line 36, in __init__
    self.count: int = count or len(self.registers)
                               ~~~^^^^^^^^^^^^^^^^
TypeError: object of type 'int' has no len()

Pepekliste, 2025-02-07:
Yep, same here.

ollila, 2025-02-08:
Looks to be quite straightforward fix to wrap output to list. I tested it with only one variable all outputs should be fixed. 

From this:

`await self._modbus.async_pb_call(
    self._unit_id,
    CTS602HoldingRegisters.central_heat_supply_min,
    output,
    "write_registers",
)
`
To this
`await self._modbus.async_pb_call(
    self._unit_id,
    CTS602HoldingRegisters.central_heat_supply_min,
    [output], 
    "write_registers",
)
`
seemed to fix it. I will try to test more later today, this would need to be changed to all outputs.

ollila, 2025-02-09:
I have now created PR for the fix. I did not explicitly test all writes but tested something from output, mode and value categories.

delinend, 2025-02-09:
> I have now created PR for the fix. I did not explicitly test all writes but tested something from output, mode and value categories.

Great, thanks :-) Can't wait for a fix.

delinend, 2025-02-14:
> Me too: CompactS Firmware: 3.6.57 Hardware: 96 , Nilan integr. 1.2.19, HA 2025.2. Change in file device.py from output to [output] no effect, the same for value or mode, or all of them. The same for output - (output) Going back HA to 2025.1. - no effect. :-(

Try the "Pull request" from @ollila. It works fine for me :-) [https://github.com/veista/nilan/pull/172](url)

jiripatka, 2025-02-14:
Thanks, manual work works :-)

closed 2025-02-23 by veista