veista/nilan #186: Home Assistant OS 2025.9.3 - Unknown error occurred

AdamGrumlik, 2025-09-17:
Hi, integration on Home Assistant OS 2025.9.3 tells me Unknown error occurred see screen. Version 1.2.22

[image]

marchello83, 2025-09-20:
The problem comes from the latest pymodbus 3.11.x.
Methods like read_holding_registers no longer accept slave= or unit= arguments.
The correct parameter is now device_id=.

For example, in config_flow.py you currently have:

`result = await client.read_holding_registers(
    CTS602HoldingRegisters.control_type, count=1, unit=int(unit_id)
)`

This needs to be changed to:

`result = await client.read_holding_registers(
    CTS602HoldingRegisters.control_type, count=1, device_id=int(unit_id)
)`

The same change (unit= → device_id=) should be applied everywhere in the integration where Modbus calls are made (read_holding_registers, read_input_registers, write_register(s), read_coils, read_discrete_inputs, write_coil(s)).

After replacing all of these, the integration works again with pymodbus ≥3.11.

flyingmoro, 2025-09-25:
i can confirm marchello83's comment. And you actually only need to change this one line in the config_flow file, then it already works.